arduino/arduino-pro-ide #364: Arduino Digispark support

silvioprog, 2020-09-28:
It would be nice to add [Arduino Digispark](https://github.com/digistump/DigistumpArduino) support (ATtiny85-based board).

When we try to upload the firmware (just a blink example), it raises the following error:

```
Error: No ports selected. Please select a port.
```

however, when we try to configure the port, it shows only `NO PORTS DISCOVERED`, as following:

[image]

ubidefeo, 2020-09-29:
@per1234 I saw you marked it as a bug, why so?
It could be that the driver installation didn't go through and I also saw in their troubleshooting and wiki that once the board is configured as a USB device it will only be visible for 5 seconds before it starts acting as a peripheral.
Might that be one of the issues?

per1234, 2020-09-29:
>why so?

In the case of DigiSpark is a vUSB device, The micronucleus uploader doesn't use `serial.port`, so Arduino Pro IDE's requirement of having a serial port selected is unnecessary.

This requirement means the only way to upload to a pure USB device is to select some random serial port, even though it won't be used for the upload. Some people might not even have a serial port available for this purpose.

A workaround the platform author to get around this limitation might be to not define the `upload.protocol` property, so the "always Upload Using Programmer" behavior is used (which doesn't require a serial port), but then you might have issues of the user needing to have the correct **Tools > Programmer** selection. Anyway, Pro IDE also doesn't support the "always Upload Using Programmer" behavior currently (https://github.com/arduino/arduino-pro-ide/issues/309).

The Arduino IDE can handle this sort of use case fine, so it's certainly achievable.

ubidefeo, 2020-09-29:
@per1234 
We need to make sure this use case is covered, we'll potentially have it with other HID or DFU based boards.
I'll put it in our task-list, thank you for chiming in 👍

ubidefeo, 2020-09-29:
@per1234 I have been researching this case.
I don't have this board, but tried to upload using the Java IDE anyway to see the command and any possible error.
I get an error I cannot recognise 
`Error while uploading: missing 'program.params.verbose' configuration parameter`

from the CLI, running `arduino-cli compile -b digistump:avr:digispark-tiny -u -P micronucleusprog` I also go nowhere and get
`Error during Upload: programming error: recipe not found 'program.pattern'`

Maybe there's something wrong with the upload recipe in the platform definition?

per1234, 2020-09-29:
>I don't have this board

Same here.
>I get an error I cannot recognise
Error while uploading: missing 'program.params.verbose' configuration parameter

This is because the platform is not designed for use with "Upload Using Programmer". I don't know why they added that "micronucleus" programmer because it can't be used.

You need to do a standard "Upload":
http://digistump.com/wiki/digispark/tutorials/connecting#using_the_digispark_with_the_arduino_ide:~:text=You%20do%20not%20need%20to%20plug%20in%20your%20Digispark%20before%20invoking%20upload
>Hit the upload button. The bottom status box will now ask you to plug in your Digispark - at this point you need to plug it in - or unplug and replug it.
>
>You'll see the upload progress and then it will immediately run your code on the Digispark.

>from the CLI, running arduino-cli compile -b digistump:avr:digispark-tiny -u -P micronucleusprog I also go nowhere and get
Error during Upload: programming error: recipe not found 'program.pattern'

Remove the `-P micronucleusprog` from the command.

>Maybe there's something wrong with the upload recipe in the platform definition?

`upload` is fine. It's the missing definition for the `program` action. In the time the Digispark boards platform was written, you had access to all programmers of all platforms and the platform.txt of the boards platform of the selected programmer was used rather than the digispark platform.txt, so it would get the `program` action definition from the programmer's platform. That behavior was changed in Arduino IDE 1.8.13 and now each platform must provide definitions for all supported programmers.

ubidefeo, 2020-09-29:
@per1234 
got it. it works with CLI (as in "it's waiting for me to plug").
so... we need to relax the port selection policy.
we'll get to it in the next sprint or two, should be doable without much hassle

kittaakos, 2020-10-21:
For the record; to properly support this use-case in IDE2, we have to remove all premature checks on the IDE2 side before giving back the control to the CLI to verify/upload. This means we cannot warn the user if a board is not selected and the user verifies. We also cannot notify the user if she wants to upload and no board/port is picked.

Current state:
1. Verify/Upload without a selected board:
[image]

2. Upload without a selected port:
[image]

kittaakos, 2020-10-21:
For the record; to properly support this use-case in the new IDE, we have to remove all premature checks on the IDE side before giving back the control to the CLI to verify/upload. This means we cannot warn the user if a board is not selected and the user verifies. We also cannot notify the user if she wants to upload and no board/port is picked.

Current state:
1. Verify/Upload without a selected board:
[image]

2. Upload without a selected port:
[image]

kittaakos, 2020-10-21:
Proposed changes:

1. Verify without a board selected:
[image]

2. Verify with a selected board but without the corresponding core installed:
[image]

As one can see 👆, we cannot distinguish between the "no board was selected" and "the core is not installed for the selected board" cases. Both results in a no FQBN was set error.

3. Uploading to a board that is not connected (no port was set):
[image]

4. Upload using a programmer when no programmer is selected:
[image]
[image]

This is tricky, when no programmer is selected, the upload using programmer will trigger an upload. There [is no dedicated gRPC API for upload using programmer](https://github.com/arduino/arduino-cli/blob/a9c0a97c50093539f0e4805e7475aeb1584b1d8a/rpc/commands/upload.proto#L24-L49), this means, if the user does not set a programmer but performs upload using programmer, it will result in a common upload call on the CLI.
(CC: @ubidefeo)

5. Burn bootloader without a programmer set:
[image]
[image]

kittaakos, 2020-10-26:
> This is tricky, when no programmer is selected, the upload using programmer will trigger an upload. There [is no dedicated gRPC API for upload using [...]

I did another iteration with https://github.com/arduino/arduino-cli/pull/1045 and it works as expected:
 - I have selected my Zero board,
 - the core was installed,
 - the FQBN was set,
 - serial port was set,
 - programmer was **not** set

I see the expected error, instead of the silent upload.

[image]

closed 2020-11-06 by kittaakos

silvioprog, 2020-12-06:
Hey dudes, is there already any nightly build containing support for Digispark board? 😅 Thank you!

kittaakos, 2020-12-07:
@silvanocerza, could you try with the latest nightly build?

kittaakos, 2020-12-07:
> could you try with the latest nightly build?

@silvioprog

silvioprog, 2020-12-07:
@kittaakos, I have been trying the latest build  (`Version: 0.1.2-nightly.20201207 / CLI Version: 0.13.0 alpha [27381c5]`), but got another problem I'm going to show below.

The full steps used to try the [Arduino Digispark ATtiny85](https://digistump.com/wiki/digispark/tutorials/connecting):

1. Click to `File | Open CLI Configuration` menu;
2. Put the URL `http://digistump.com/package_digistump_index.json` to `additional_urls` key in `board_manager` section and save the YAML file;
3. Lookup the Digispark board at `Tools | Boards manager (Ctrl + Shift + M)` and install it;
4. Use the `Select Other Board & Port` option in the toolbar, search for 'Digispark' and select the `Digispark (Default - 16.5mhz)`.
5. Open the `Blink` example, compile and try to upload it. (remember to add `#define LED_BUILTIN 1` on top of the file)

At this step, the IDE reported the following errors:

```
Error: Request upload failed with message: 2 UNKNOWN: exit status 1
```
```
Error: Request upload failed with message: 2 UNKNOWN: uploading error: cannot execute upload tool: exec: "{runtime.tools.micronucleus.path}/launcher": file does not exist
```

Additional information:

- Yesterday, I tried the [PlatformIO](https://platformio.org) in VSCode Insiders and uploaded the blink example to the Digispark board properly.
- Today, I accidentally short-circuited my Digispark board (😕), but I'm waiting to receive a new one to try all the tests again (🙂).

kittaakos, 2020-12-07:
Thanks for getting back with the details, @silvioprog. I wanted to verify it with the CLI, I do not have the board, but I could not even compile the sketch.

```
akos.kitta@Akoss-MacBook-Pro Desktop % rm -rf ~/Library/Arduino15                                                                        
akos.kitta@Akoss-MacBook-Pro Desktop % rm -rf ~/Documents/Arduino/hardware                                                               
akos.kitta@Akoss-MacBook-Pro Desktop % ./arduino-cli version --format json                                                               
{
  "Application": "arduino-cli",
  "VersionString": "0.14.0-rc1",
  "Commit": "7d4baa9",
  "Status": "alpha",
  "Date": "2020-12-03T17:24:59Z"
}
akos.kitta@Akoss-MacBook-Pro Desktop % ./arduino-cli core update-index --additional-urls http://digistump.com/package_digistump_index.json --format json
akos.kitta@Akoss-MacBook-Pro Desktop % ./arduino-cli core install digistump:avr --additional-urls http://digistump.com/package_digistump_index.json --format json
akos.kitta@Akoss-MacBook-Pro Desktop % cat ~/Documents/Arduino/DigisparkBlink/DigisparkBlink.ino 

#define LED_BUILTIN 1

void setup() {
  pinMode(LED_BUILTIN, OUTPUT);
}

void loop() {
  digitalWrite(LED_BUILTIN, HIGH);   // turn the LED on (HIGH is the voltage level)
  delay(1000);                       // wait for a second
  digitalWrite(LED_BUILTIN, LOW);    // turn the LED off by making the voltage LOW
  delay(1000);                       // wait for a second
}
akos.kitta@Akoss-MacBook-Pro Desktop % ./arduino-cli board listall  --additional-urls http://digistump.com/package_digistump_index.json --format json
{
  "boards": [
    {
      "name": "Digispark Pro (Default 16 Mhz)",
      "FQBN": "digistump:avr:digispark-pro"
    },
    {
      "name": "Digispark Pro (16 Mhz) (32 byte buffer)",
      "FQBN": "digistump:avr:digispark-pro32"
    },
    {
      "name": "Digispark Pro (16 Mhz) (64 byte buffer)",
      "FQBN": "digistump:avr:digispark-pro64"
    },
    {
      "name": "Digispark (16mhz - No USB)",
      "FQBN": "digistump:avr:digispark-tiny16"
    },
    {
      "name": "Digispark (8mhz - No USB)",
      "FQBN": "digistump:avr:digispark-tiny8"
    },
    {
      "name": "Digispark (1mhz - No USB)",
      "FQBN": "digistump:avr:digispark-tiny1"
    },
    {
      "name": "Digispark (Default - 16.5mhz)",
      "FQBN": "digistump:avr:digispark-tiny"
    }
  ]
}
akos.kitta@Akoss-MacBook-Pro Desktop % ./arduino-cli compile -b digistump:avr:digispark-tiny ~/Documents/Arduino/DigisparkBlink --additional-urls http://digistump.com/package_digistump_index.json --format json 
{
  "compiler_out": "",
  "compiler_err": "",
  "builder_result": {
    "build_path": "/var/folders/k3/d2fkvv1j16v3_rz93k7f74180000gn/T/arduino-sketch-F389360393BFFB04E3EC135340A2EF23"
  },
  "success": false
}
akos.kitta@Akoss-MacBook-Pro Desktop % ./arduino-cli compile -b digistump:avr:digispark-tiny ~/Documents/Arduino/DigisparkBlink --additional-urls http://digistump.com/package_digistump_index.json             

Error during build: fork/exec /Users/akos.kitta/Library/Arduino15/packages/arduino/tools/avr-gcc/4.8.1-arduino5/bin/avr-g++: bad CPU type in executable
akos.kitta@Akoss-MacBook-Pro Desktop % 

```

kittaakos, 2020-12-09:
> At this step, the IDE reported the following errors:
>
> ```
> Error: Request upload failed with message: 2 UNKNOWN: exit status 1
> ```
>
> ```
> Error: Request upload failed with message: 2 UNKNOWN: uploading [...]

@ubidefeo, could you please help to verify the compilation with the CLI? The IDE cannot do much with the `cannot execute upload tool: exec: "{runtime.tools.micronucleus.path}/launcher": file does not exist` error. Unless there is a gRPC API misuse on the new IDE side 😅 Thank you!