arduino/arduino-pro-ide #365: Add support to open Arduino Libraries (=folders)

tuxedo0801, 2020-10-01:
**Is your feature request related to a problem? Please describe.**
If I'm an Arduino Library deeloper, I'm not able to open a lilbrary and edit it. 

**Describe the solution you'd like**
Opening a project filters for sketches only. Would be great if IDE allows to open folders (like VS Code allows) - maybe by switching filter from "sketch" to "folder" - 

**Describe alternatives you've considered**
Use VS Code instead of Arduino Pro IDE :-( --> Maintain two different IDEs for one attempt: Develop Arduino stuff..

**Additional context**
-

tuxedo0801, 2020-10-01:
Semms to be related to https://github.com/arduino/arduino-pro-ide/issues/317

per1234, 2020-10-01:
Just to link related discussions together, I'm adding a reference to the discussion about adding this capability to the Arduino IDE: https://github.com/arduino/Arduino/issues/3512