arduino/arduino-pro-ide #368: error when #include "Arduino.h" on macOS

sstaub, 2020-10-17:
**Describe the bug**
see screenshot, I got an error (but I can compile) when I include Arduino.h

```
In included file: Unsupported architecture
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.15.sdk/usr/include/sys/cdefs.h:807:2:
note: error occurred hereclang(pp_hash_error)
```

I know that Including Arduino.h is not needed, but necessary when use in platformIO.
Maybe a general problem with Xcode because GCC is no more supported on Xcode 12.

**To Reproduce**
Create a file with `#include "Arduino.h"`

**Expected behavior**
there should no error

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: macOS 15.7
 - Xcode 12
 - Version: 0.1.2 nightly

**Additional context**
Add any other context about the problem here.

ubidefeo, 2021-02-17:
@sstaub 
this is not an error for the IDE, the sketch compiles fine and that header is automatically added to any Arduino sketch compilation process.
I can imagine the highlight as an error bothers you, but it should not be a big concern

per1234, 2021-02-19:
Hi @sstaub . Thanks so much for taking the time to report this!

This should now be working in [the new 0.1.4 release of Arduino Pro IDE](https://github.com/arduino/arduino-pro-ide/releases/tag/0.1.4).

If you try it out and find that the problem is still occurring, feel free to comment here and we'll investigate.

closed 2021-02-19 by per1234