arduino/arduino-pro-ide #369: Always "no board selected"

RussellAult, 2020-10-18:
**Describe the bug**
Every time I open the IDE "no board selected" appears at the top of the screen, even if a board was selected when the sketch was saved. I then have to go in through the "Select Other Board & Port" function and search for the board.

**To Reproduce**
1. Create a new sketch.
2. Select a board for the sketch.
3. Save the sketch.
4. Close the IDE.
5. Open the IDE (which re-opens the previously-saved sketch).

**Expected behaviour**
Board that was selected in step 2 is still selected.

**Actual behaviour**
`no board selected` appears at the top of the screen, and warnings appear about any libraries that are part of the selected Core being missing.

**Environment**
- OS: Windows 10 Pro
- Version: 0.1.2

**Additional context**

- I'm using the "ATtiny25/45/85 (No bootloader)" board from ATtinyCore with a USBtinyISP-based programmer; no ports are detected, and once selected the board appears in the top bar with a red "X" beside it (although my understanding is that both of these are normal when using a programmer?).
- Thankfully, after re-selecting the board, all of the other board-based settings (chip, clock, programmer type, etc.) appear to be their previously-saved values (and not defaults).
- For what it's worth I've had no trouble compiling or uploading, it's just annoying to have to re-find the board every time.

Dave528, 2020-11-11:
This problem seems to be related to the IDE not detecting serial com ports at startup.  I have the same issue with Windows 7 x64.
When the IDE starts the previously selected board is still shown, but with a red 'X' indicating that the board was not found.  I have to manually select 'Select other board & port' from the drop down and re-select the com port.
The same problem also occurs if the board is unplugged from the USB and later re-attached.  The com port is simply not found and selected.

RussellAult, 2020-11-11:
@Dave528 That sounds like it might be a whole separate issue; in my case the previously-selected board is never shown.

ubidefeo, 2021-02-17:
The last version supports remembering the last used board

closed 2021-02-17 by ubidefeo