arduino/arduino-pro-ide #370: Support customization of board selection drop-down

RussellAult, 2020-10-18:
**Is your feature request related to a problem? Please describe.**
When editing sketches "offline" (or, as I understand it, when using a bootloader-less chip and a programmer) the only thing that appears in the board selection drop-down is "Select Other Board & Port". Even if I only ever used a couple of boards, I always have to go in and search for them (which makes #369 a lot more annoying).

**Describe the solution you'd like**
It would be great if there was a way to populate the drop-down with frequently-used boards. I could see this being populated manually (either as part of preferences or as a "favourite" button in the board selection window), automatically (e.g. the last five boards used appear in the drop-down), or some combination of the two.

ubidefeo, 2020-10-21:
@RussellAult 
this looks like an interesting feature.
Let me throw it at our UX team and see what they think 👍

per1234, 2021-02-19:
Now that the **Tools > Board** menu has been added, it might be worth studying what has already been done in this space for the classic Arduino IDE:
- https://github.com/arduino/Arduino/pull/10816
- https://github.com/arduino/Arduino/pull/8607

ubidefeo, 2021-02-19:
@per1234 
I really like the idea of having recently used boards.
Sometimes you just want to make sure you can compile for something you're developing on as a target without having it attached