arduino/arduino-pro-ide #371: Use separate tooltip for status icon in board selection drop-down

RussellAult, 2020-10-18:
**Is your feature request related to a problem? Please describe.**
This is pretty minor (and perhaps more difficult to achieve than it's worth, especially in the context of what I'm sure is inevitably-forthcoming documentation), but as it stands I have absolutely no clue what the red "X" in the board selection drop-down actually means, and the tooltip when I hover over the "X" is the same as for the rest of the drop-down (i.e. the un-trunkated name of the board).

**Describe the solution you'd like**
A separate tooltip for the status icon describing its meaning would be great!

**Describe alternatives you've considered**
As mentioned above, presumably somewhere down the road the status icon's meaning will be made clear in documentation, but even in that context I think the tooltip would still be useful.

ubidefeo, 2020-10-21:
@RussellAult 
if you are referring to this
[image]

I believe it's pretty self-explanatory that it means "board is not connected". A red cross can't have much more meaning than that.
When you expand the menu you will see that it is, in fact, not available in the list.
This is probably much more easily classified as a UI bug, because a previously selected board is still listed as associated to the current sketch while the board is not available to the system.
I think we'll move forward with fixing the bug, rather than adding tooltip functionalities.
Thank you for bringing this to our attention :)

RussellAult, 2020-10-21:
It may seem self-explanatory, but only for users who have ever seen something else (which doesn't include me). The only chip I've used the Arduino Pro IDE with (so far) is a bootloader-less ATtiny, so I have no frame of reference for that X since it is always there. Up until now I haven't been sure if it means "board not connected" or "board with bootloader not connected" (which is, I believe, the correct answer) or even "errors present in the source code will prevent the current sketch from compiling" (clang has been very unhappy with me, too).

Looking at it another way, I guess I'm wondering, on a fundamental level, what the status indicator is really trying to say? A big red X should be trying to tell me that something is seriously wrong, but in the case of bootloader-less boards it's the only indicator that will show up, even if everything is working perfectly. Perhaps the solution isn't (just?) a tooltip, but instead status indicators that are more responsive the context of the chip being dealt with? I'm not sure. All I know is that I've been confused by that red X since I first opened the software...

ubidefeo, 2020-10-21:
@RussellAult 
UX-wise those are all valid points.
I'm happy to bring them to the table at our next UX/UI conversation 👍