arduino/arduino-pro-ide #376: I installed a library, and It's not showing up.

Void-Overflow, 2020-10-27:
Hello, I recently downloaded the pro ide, and I tried downloading the Adafruit_bmp085.h library. I downloaded it, and when I try to install it it doesn't find it. I'm not sure what the issue is. Also, how would I go about installing a zip library for the ide. I have a tinygps++ library that I also wish to install, and I'm  not sure how to do it through the pro ide. Anyway, here is the begginging of my code: 

```cpp
#include <Arduino.h>
#include <Wire.h>
#include <Adafruit_BMP085.h>
#include <TinyGPS++.h>
#include <SoftwareSerial.h>

static const int RXPin = 10, TXPin = 11;
static const uint32_t GPSBaud = 115200;

TinyGPSPlus gps;
SoftwareSerial ss(RXPin, TXPin);
Adafruit_BMP085 bmp;
```

Void-Overflow, 2020-10-27:
Also, it seems to be happening to not just this library, but other libraries I've installed. Such as Liquid Crystal.

ubidefeo, 2020-10-27:
@Void-Overflow 
if you don't want to use the Library Manager (maybe a specific library or a version of it are not in the main branch) you can definitely download it as zip and expand it.
As best practice remove the `-master` or `-main` suffix from the folder name, and then move/copy it into the `libraries` folder inside your sketchbook.
Restart the IDE and you'll be good to go (if the library is correctly structured, of course)

Void-Overflow, 2020-10-27:
Hi ubidefeo, where is the sketchbook for the pro IDE? I don't see it in the downloads folder. Is it in the same directory as the Arduino IDE?

Void-Overflow, 2020-10-27:
Another question I have is, is it possible to set a default workspace in the settings.json? If so, how would I structure it?

Void-Overflow, 2020-10-27:
One last thing I noticed is that, when I go under examples, the library does come up, but when I go to it that example sketch also has the same issue where the library is not recognized.

ubidefeo, 2020-10-27:
@Void-Overflow 
in general it's in the user folder.
You can create a new sketch, save it and choose from the menu `Sketch`>`Show Sketch Folder`.
The `libraries` colder should be at the root of the SketchBook folder, which usually is called `Arduino`.
On WIndows, for instance, is in `Documents`>`Arduino`.
If such folder is not there just create it yourself

ubidefeo, 2020-10-27:
@Void-Overflow 

your library is not correctly installed.
Please check the installation instructions on the library's github or forum (if available)

Void-Overflow, 2020-10-27:
Hi, I tried putting the libraries into the project directory, and now I get a compilation error : Compilation error: Error: 2 UNKNOWN: exit status 1

Void-Overflow, 2020-10-28:
This is the compilation: 
```
Using board 'mega' from platform in folder: C:\Users\dehla\AppData\Local\arduino15\packages\arduino\hardware\avr\1.8.3
Using core 'arduino' from platform in folder: C:\Users\dehla\AppData\Local\arduino15\packages\arduino\hardware\avr\1.8.3
Detecting libraries used...
"C:\\Users\\dehla\\AppData\\Local\\arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR "-IC:\\Users\\dehla\\AppData\\Local\\arduino15\\packages\\arduino\\hardware\\avr\\1.8.3\\cores\\arduino" "-IC:\\Users\\dehla\\AppData\\Local\\arduino15\\packages\\arduino\\hardware\\avr\\1.8.3\\variants\\mega" "C:\\Users\\dehla\\AppData\\Local\\Temp\\arduino-sketch-B1074CFF9593B0633248A6EA93039253\\sketch\\sketch_oct27a.ino.cpp" -o nul
Alternatives for Wire.h: [Wire@1.0]
ResolveLibrary(Wire.h)
  -> candidates: [Wire@1.0]
"C:\\Users\\dehla\\AppData\\Local\\arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR "-IC:\\Users\\dehla\\AppData\\Local\\arduino15\\packages\\arduino\\hardware\\avr\\1.8.3\\cores\\arduino" "-IC:\\Users\\dehla\\AppData\\Local\\arduino15\\packages\\arduino\\hardware\\avr\\1.8.3\\variants\\mega" "-IC:\\Users\\dehla\\AppData\\Local\\arduino15\\packages\\arduino\\hardware\\avr\\1.8.3\\libraries\\Wire\\src" "C:\\Users\\dehla\\AppData\\Local\\Temp\\arduino-sketch-B1074CFF9593B0633248A6EA93039253\\sketch\\sketch_oct27a.ino.cpp" -o nul
Alternatives for Adafruit_BMP085.h: [Adafruit_BMP085_Library@1.1.0]
ResolveLibrary(Adafruit_BMP085.h)
  -> candidates: [Adafruit_BMP085_Library@1.1.0]
"C:\\Users\\dehla\\AppData\\Local\\arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR "-IC:\\Users\\dehla\\AppData\\Local\\arduino15\\packages\\arduino\\hardware\\avr\\1.8.3\\cores\\arduino" "-IC:\\Users\\dehla\\AppData\\Local\\arduino15\\packages\\arduino\\hardware\\avr\\1.8.3\\variants\\mega" "-IC:\\Users\\dehla\\AppData\\Local\\arduino15\\packages\\arduino\\hardware\\avr\\1.8.3\\libraries\\Wire\\src" "-IC:\\Users\\dehla\\OneDrive\\Documents\\Arduino\\libraries\\Adafruit_BMP085_Library" "C:\\Users\\dehla\\AppData\\Local\\Temp\\arduino-sketch-B1074CFF9593B0633248A6EA93039253\\sketch\\sketch_oct27a.ino.cpp" -o nul
Alternatives for TinyGPS++.h: []
ResolveLibrary(TinyGPS++.h)
  -> candidates: []
Using library Wire at version 1.0 in folder: C:\Users\dehla\AppData\Local\arduino15\packages\arduino\hardware\avr\1.8.3\libraries\Wire 
Using library Adafruit_BMP085_Library at version 1.1.0 in folder: C:\Users\dehla\OneDrive\Documents\Arduino\libraries\Adafruit_BMP085_Library 
Compilation error: Error: 2 UNKNOWN: exit status 1
```

ubidefeo, 2020-10-28:
@Void-Overflow 
the libraries should not be in the "project directory" (which I guess is your sketch folder where your `.ino` file is).
Libraries have to be in the `libraries` folder inside the Arduino sketchbook folder (which is where all your Arduino sketches are saved by default when you create a new sketch and save it)

The defaults are

Mac OS: ~/Documents/Arduino/
Windows: \Users\YOURNAME\Documents\Arduino\
Linux: ~/Arduino/

the sketchbook folder can be changed by setting some preferences, but for the sake of testing if this library works just use the default one.
Inside this folder there should be a `libraries` folder, but if there isn't one (no libraries installed) just create it and copy the downloaded folder into that one.

Void-Overflow, 2020-11-01:
Hey, thanks for your help I managed to fix the issue!

closed 2020-11-01 by Void-Overflow

ubidefeo, 2020-11-01:
my pleasure, @Void-Overflow 👍