arduino/arduino-pro-ide #377: Adafruit Trinket M0 won't persist port

rkincaid, 2020-10-27:
**System**
MacBook (Retina, 12-inch, 2017)
MacOS 10.15.7 (Catalina), 12
Pro IDE Version 0.1.2 (0.1.2)
Adafruit Trinket M0

I can get the IDE to properly compile my sketch AND transfer correctly to the Trinket, but after transfer it loses the port setting and requires I manually redefine the port setting again in the "select other board & port" dropdown. Additionally, it appears to recognize the Trinket incorrectly as an "Adafruit pIRKey" as this entry is always associated with the correct USB port and I never added it. If I manually specify the Trinket and USB port, it works once, but then the Trinket loses the port and the pIRKey definition remains. Also, it never persists my Trinket setting (regardless of port) once I close the IDE. The behavior seems to be that it insists on seeing the board as a pIRKey, but that's just a guess on my part.

I'm not sure if this has anything to do with it, but I've noticed on my Mac that once the sketch is installed and the board resets, the USB connection temporarily disconnects and reconnects. And I get the red X in the board combobox at the top of the screen. Perhaps this temporary disconnect is confusing the board settings?

QUESTION: Is there anyway to remove boards from the board/port dropdown once they have been associated?

Also, FWIW, the standard IDE does just fine with the same sketch and persists settings correctly and recovers from the temporary board disconnect.

**To Reproduce**
Steps to reproduce the behavior:
1. Install IDE, Trinket support (per Adafruit instructions), and required sketch libraries.
2. Plug in Trinket.
3. Observe that board settings automatically pick up an entry for pIRKey with the correct port we need for the Trinket
4. Manually select the build for the Trinket M0 and correct board.
5. Compile and upload successfully.
6. Immediately loses the port setting and won't upload correctly until you repeat #4
7. Close IDE, reopen and it's as if the Trinket was never there. 

**Expected behavior**
Trinket and port settings should persist like the standard IDE

rkincaid, 2020-10-27:
UPDATE with workaround:

The following seems to work around the problem, but it still seems like something isn't right in order for this to matter...

1. Edit /Users/robertkincaid/Library/Arduino15/packages/adafruit/hardware/samd/1.6.3/boards.txt
2. remove he entry for "Adafruit PIRkey (SAMD21)"
3. save boards.txt

ubidefeo, 2021-02-17:
hi @rkincaid 
as we groom our older issues I found this one and tested.
It indeed suddenly picks up the board as a pIRKey.
This might be due to the VID:PID pair being shared across Adafruit's boards (I have not investigated), hence you should report this to their forums or GH repos for their SAMD boards.

rkincaid, 2021-02-17:
Thanks for the followup. I would note however that this does not seem to occur with the non-pro IDE or with PlatformIO. I ended up moving to PlatformIO/Microsoft Code.

ubidefeo, 2021-02-17:
thank you for replying so promptly, @rkincaid 

@per1234 can you look into this?
I'm sure you can quickly figure out why this happens :)

per1234, 2021-02-17:
>This might be due to the VID:PID pair being shared across Adafruit's boards

That's correct. Here's the VID/PID definition for the Trinket M0:
https://github.com/adafruit/ArduinoCore-samd/blob/1.6.5/boards.txt#L290-L293
```
adafruit_trinket_m0.vid.0=0x239A
adafruit_trinket_m0.pid.0=0x801E
adafruit_trinket_m0.vid.1=0x239A
adafruit_trinket_m0.pid.1=0x001E
```
and here for the pIRkey:
https://github.com/adafruit/ArduinoCore-samd/blob/1.6.5/boards.txt#L433-L436
```
adafruit_pirkey.vid.0=0x239A
adafruit_pirkey.pid.0=0x801E
adafruit_pirkey.vid.1=0x239A
adafruit_pirkey.pid.1=0x001E
```
So either one will match the automatic board detection. I don't own a Trinket M0, but I was able to reproduce the issue with the native port of my Zero by just adding another board definition to the Arduino SAMD Boards boards.txt with the same VID/PID definition but different board ID and name.

> I'm not sure if this has anything to do with it, but I've noticed on my Mac that once the sketch is installed and the board resets, the USB connection temporarily disconnects and reconnects. And I [...]

That's exactly it. The board resets at the start of the upload and the bootloader runs, which has a different VID/PID, then the upload finishes and the application runs, which makes the standard VID/PID appear on the system. This triggers the Pro IDE/Arduino CLI's automatic board detection to happen and it seems to pick one of the two boards at random to associate with the port. You can actually reproduce the issue simply by pressing the reset button on the board, no upload needed.

>this does not seem to occur with the non-pro IDE

That's because it doesn't have the automatic board detection feature. It just uses whichever board you manually selected from the **Tools > Board** menu.

>remove he entry for "Adafruit PIRkey (SAMD21)"

I think that's the best provisional workaround for anyone encountering this issue.

ubidefeo, 2021-02-18:
thank you, @per1234