arduino/arduino-pro-ide #378: Always getting Error 2

MichaelRLebert, 2020-10-28:
Sorry for the inconvienience and maybe its obvious for must of you what to do. When I try to compile a very simple LED on/off example I get the following output response:
Compiling...
{
  "sketchUri": "file:///c%3A/Users/Michael/AppData/Local/Temp/.arduinoProIDE-unsaved2020928-20924-iz94r5.yrag9/sketch_oct28a/sketch_oct28a.ino",
  "fqbn": "arduino:mbed:envie_m4",
  "optimizeForDebug": false
}
--------------------------
Using board 'envie_m4' from platform in folder: C:\Users\Michael\AppData\Local\Arduino15\packages\arduino\hardware\mbed\1.3.0
Using core 'arduino' from platform in folder: C:\Users\Michael\AppData\Local\Arduino15\packages\arduino\hardware\mbed\1.3.0
Detecting libraries used...
/bin/arm-none-eabi-g++ -c -w -g -Os -nostdlib "@C:\\Users\\Michael\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed\\1.3.0\\variants\\PORTENTA_H7_M4/defines.txt" "@C:\\Users\\Michael\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed\\1.3.0\\variants\\PORTENTA_H7_M4/cxxflags.txt" -mcpu=cortex-m4 -mfloat-abi=softfp -mfpu=fpv4-sp-d16 -w -x c++ -E -CC -DARDUINO=10607 -DARDUINO_PORTENTA_H7_M4 -DARDUINO_ARCH_MBED -DARDUINO_LIBRARY_DISCOVERY_PHASE=1 "-IC:\\Users\\Michael\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed\\1.3.0\\cores\\arduino" "-IC:\\Users\\Michael\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed\\1.3.0\\variants\\PORTENTA_H7_M4" "-IC:\\Users\\Michael\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed\\1.3.0\\cores\\arduino/api/deprecated" "-iprefixC:\\Users\\Michael\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed\\1.3.0\\cores\\arduino" "@C:\\Users\\Michael\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed\\1.3.0\\variants\\PORTENTA_H7_M4/includes.txt" "C:\\Users\\Michael\\AppData\\Local\\Temp\\arduino-sketch-F3453CD5BFE6D0190CC1EC46CEABAD7A\\sketch\\sketch_oct28a.ino.cpp" -o nul
Compilation error: Error: 2 UNKNOWN: exec: "/bin/arm-none-eabi-g++": file does not exist

I did search all obvious sources, but no luck. Could you please point me in the right direction?

ubidefeo, 2020-10-29:
hi @MichaelRLebert 
I'm investigating the matter, will get back to you asap :)

ubidefeo, 2020-10-30:
hey @MichaelRLebert 
I ended up having this issue, which was due to deprecation of the `arduino-beta:mbed` beta core you installed in order to support Portenta.
The supporting files are now part of `arduino:mbed`, so I suggest you remove that index and the board support and update `arduino:mbed`.

It should get everything back in order
u.

ubidefeo, 2021-02-17:
closed for lack of feedback

closed 2021-02-17 by ubidefeo