arduino/arduino-pro-ide #379: SAVE NOT WORKING

DeeEmm, 2020-10-30:
**Describe the bug**
Save not working

**To Reproduce**
Steps to reproduce the behaviour:
1. use editor
2. save work at regular intervals (every edit / few mins / force of habit)
3. everything apparently fine, no visual indicators that save not working
4. try to verify code and note that verify button not working
5. restart IDE and find that no changes have been saved for past 9 hours
6. check in temp folder in case it is a recovery file issue
7. no interim save data found in temp folder
8. no way to recover
9. cry at an entire days lost work

**Expected behaviour**
Editor to save when save action called
OR
editor to alert user that save action failed

**Desktop (please complete the following information):**
 - OSX 10.15.7
 - ArduinoPro Version 0.1.2

**Additional context**
- I turned USB hub off when I left computer (arduino plugged into hub)
- Computer went to sleep (possibly not relevant)
- Have had openandsavepanelservice issues in other apps (chrome) (possibly relevant)

DeeEmm, 2020-10-30:
This may be relevant:

https://discussions.apple.com/thread/251354974?answerId=253763949022#253763949022

I suspect that it is an openandsavepanelservice related issue and that the exception is not caught by Pro-IDE, which allows the user to continue to work on the project unaware that the save function is not working (both the save dialog and when accessed directly via shortcut keys 'CMD-S').

It would be better if the exception was caught and the app locked up - basically what Chrome does. At least the user is then made aware of the fact that there is an issue.

kittaakos, 2020-10-30:
> I suspect that it is an openandsavepanelservice related issue

Most likely no.

Do you happen to remember whether the status bar was yellow or not? If it was yellow, unfortunately, you have hit an offline issue that used to happen after waking up the OS from sleep on macOS. The bug has been fixed and available part of the [nightly](https://downloads.arduino.cc/arduino-pro-ide/nightly/CHANGELOG.txt):

>  - Capture and swallow unhandled SIGPIPE signal. [30136b0]

kittaakos, 2020-10-30:
> * Computer went to sleep (possibly not relevant)

Ohh, I just noticed. I am positive that was the problem. I am very very sorry about your lost work. Please get the latest nightly ASAP. Thank you!

kittaakos, 2020-10-30:
I am going to close this issue not just because it has been fixed, but the title is misleading. Let me know if you disagree.

closed 2020-10-30 by kittaakos

DeeEmm, 2020-10-30:
I cannot recall if the bar was yellow, however I notice that this morning the bar is actually yellow as predicted and if I save the document (CMD-S) the file modified date does not change, so yes I agree, it appears to be related the issue you mention.

The lost work is my bad, one of the risks of using pre-prelease software. The only problem is that now I cannot figure out how I wrote some aspects of it. It's definitely not as elegant as the first time I wrote it.  lol.

/DM

DeeEmm, 2020-10-30:
And yes, the title is more a description of a symptom than the root cause of the problem.