arduino/arduino-pro-ide #380: As a user I want to be notified by the IDE when saving the editor content has failed.

kittaakos, 2020-11-02:
This is a follow-up task of https://github.com/arduino/arduino-pro-ide/issues/379

@DeeEmm wrote [here]:
> the bar is actually yellow as predicted and if I save the document (CMD-S) the file modified date does not change

On macOS (maybe other OSs are also affected), the backend process could crash after waking up the system from sleep. This issue has been identified and fixed in the Theia IDE framework and the new IDE's nightly build also contains this fix. However, it is not always obvious to the users what has happened and what to do.
 - First off, we have to fix the dirty indicator bug in Theia: https://github.com/eclipse-theia/theia/issues/8692
 - We have to raise a toast error message to the user when saving the editor content has failed.

kittaakos, 2020-11-06:
Depends on #315.

kittaakos, 2020-11-06:
I have some updates on this issue. (Please also check the _work in progress_ screencast. I've forcefully `kill`ed the process that does the file system save.)

 - We show a warning status bar, plus the _Offline_ status.
 - We show the dirty indicator (`•`) in the editor if it contains unsaved changes.
 - We toast an error message; saving the sketch has failed.

[image]

DeeEmm, 2020-11-06:
Awesome work @kittaakos

ubidefeo, 2020-11-09:
good one, @kittaakos 

let's reword that message to make sure the user understand that it's a bug.
Also can we intercept the user closing the window/IDE and do something about it?
Just asking for a friend 👍

kittaakos, 2020-11-09:
> let's reword that message to make sure the user understand that it's a bug.

Sure. What kind of message do you want to see? Please advise.

> Also can we intercept the user closing the window/IDE and do something about it?

Sorry, I do not understand it; can you please clarify? Thank you!

ubidefeo, 2020-11-10:
>Also can we intercept the user closing the window/IDE and do something about it?
Sorry, I do not understand it; can you please clarify? Thank you!

we should prevent the user from quitting the IDE without notifying that current content will be lost.
Even better would be to handle this like Sublime Text does, it keeps files in a temporary state until properly saved, but maybe it's convoluted

kittaakos, 2020-11-10:
> Even better would be to handle this like Sublime Text does, it keeps files in a temporary state until properly saved, but maybe it's convoluted

I cannot much more agree. That is exactly how the Theia framework should behave. VS Code switched to that behavior as well. `unload` and `beforeunload` events are deprecated and do not work reliably in the browser window: https://developers.google.com/web/updates/2018/07/page-lifecycle-api#events

kittaakos, 2020-11-10:
> VS Code switched to that behavior as well.

Interesting, the VS Code guys just revived this feature in the 1.51.0 release: https://code.visualstudio.com/updates/v1_51#_prevent-accidental-close

ubidefeo, 2020-11-10:
>Interesting, the VS Code guys just revived this feature in the 1.51.0 release: https://code.visualstudio.com/updates/v1_51#_prevent-accidental-close

boom!

kittaakos, 2020-12-01:
> we should prevent the user from quitting the IDE without notifying that current content will be lost.

We will show a different dialog detail when quitting the app, and closing the app after an unrecoverable error:
[image]
[image]

> Even better would be to handle this like Sublime Text does, it keeps files in a temporary state until properly saved, but maybe it's convoluted

It's a brilliant idea but is a giant feature and should be handled by the Theia framework.

kittaakos, 2020-12-16:
It's done.

closed 2020-12-16 by kittaakos