arduino/arduino-pro-ide #381: Doesn't show any window after launch on MacOS

skrello, 2020-11-08:
**Describe the bug**
Application after launch doesn't show any window at 1 of 2 my macbooks. And after I quiet application shows error.

**To Reproduce**
Steps to reproduce the behavior:
1. Launch the application
2. No windows open.
3. Quit application.
4. See error.

**Expected behavior**
Expect to see IDE winows after launch. And no error after close app.

**Screenshots**
[image]
[image]

**Desktop (please complete the following information):**
 - OS: MacOS
 - Version: 10.15.7

**Additional context**
At work at macbook pro 2017 IDE works fine, but at home macbook pro 2018 IDE doesn't show any window.

ubidefeo, 2020-11-09:
@skrello 
could you add the IDE version you are using?

I remember seeing a startup error in one of our internal builds a long time ago, but couldn't remember which error showed up.
@kittaakos does this error say anything to you?

kittaakos, 2020-11-09:
> does this error say anything to you?

No, it does not.

@skrello, which version of the IDE do you use? What do you see when you launch the app from a terminal?
```
% /Applications/Arduino\ Pro\ IDE.app/Contents/MacOS/Arduino\ Pro\ IDE
```

Thanks!

skrello, 2020-11-15:
Version of Arduino Pro IDE - 0.1.2

Here is the terminal output:
[image]

kittaakos, 2020-11-15:
Wow, crazy. Seems like you have a DNS issue, and you cannot resolve `localhost`. Can you ping `localhost`?
On Windows it's:
```
ping -t localhost
```

Currently, I am not on my mac but it should be something like on macOS:
```
ping localhost -c 5
```

If you cannot ping `localhost,` maybe your DNS settings are messed up. Can you ping `127.0.01`? Thanks!

skrello, 2020-11-15:
mikhailo@Mike-MBP ~ % ping 127.0.0.1 -c 5
PING 127.0.0.1 (127.0.0.1): 56 data bytes
64 bytes from 127.0.0.1: icmp_seq=0 ttl=64 time=0.050 ms
64 bytes from 127.0.0.1: icmp_seq=1 ttl=64 time=0.106 ms
64 bytes from 127.0.0.1: icmp_seq=2 ttl=64 time=0.103 ms
64 bytes from 127.0.0.1: icmp_seq=3 ttl=64 time=0.103 ms
64 bytes from 127.0.0.1: icmp_seq=4 ttl=64 time=0.108 ms

--- 127.0.0.1 ping statistics ---
5 packets transmitted, 5 packets received, 0.0% packet loss
round-trip min/avg/max/stddev = 0.050/0.094/0.108/0.022 ms

skrello, 2020-11-15:
Also turned off Firewall and it din't help

kittaakos, 2020-11-15:
Thanks for the prompt reply, good to know you can ping `127.0.0.`, but can you ping it as `localhost`? It was not clear to me.

skrello, 2020-11-15:
mikhailo@Mike-MBP ~ % ping localhost
ping: cannot resolve localhost: Unknown host

kittaakos, 2020-11-15:
> mikhailo@Mike-MBP ~ % ping localhost
> ping: cannot resolve localhost: Unknown host

Thanks. It helped a lot.

kittaakos, 2020-11-16:
@skrello, I am a little bit reluctant to quickly change the IDE code and replace all `localhost` addresses with `127.0.0.1` to fix the issue on your side. Although I am positive this change will temporarily work around the problem, we do not have control over any 3rd party dependencies and libraries we use in the new IDE. The same error can happen at any time.

Would you be up for trying to figure out why your macOS cannot resolve `localhost`? I will try to assist you if you need me. What does `cat /private/etc/hosts` print on your environment? For the sake of complete transparency, here is mine output:

```bash
cat /private/etc/hosts
##
# Host Database
#
# localhost is used to configure the loopback interface
# when the system is booting.  Do not change this entry.
##
127.0.0.1	localhost
255.255.255.255	broadcasthost
::1             localhost
# Added by Docker Desktop
# To allow the same kube context to work on the host and the container:
127.0.0.1 kubernetes.docker.internal
# End of section
```

Thank you!

skrello, 2020-11-16:
My `hosts` file is empty. I filled it with the same info as yours. Unfortunately it didn't help to IDE. If you need my help, I am ready to help you resolve the issue.

kittaakos, 2020-11-16:
Great! You do not need the `127.0.0.1 kubernetes.docker.internal` entry. Can you ping `localhost` after restating your macOS?

skrello, 2020-11-16:
`hosts` contains:
```
##
# Host Database
#
# localhost is used to configure the loopback interface
# when the system is booting.  Do not change this entry.
##
127.0.0.1	localhost
255.255.255.255	broadcasthost
::1             localhost
# End of section
```

And after restart terminal output:
```
mikhailo@Mike-MBP ~ % ping localhost
ping: cannot resolve localhost: Unknown host
```

kittaakos, 2020-11-16:
Can you check the line endings in the file? According to this [thread](https://stackoverflow.com/questions/19732360/ping-cannot-resolve-localhost-unknown-host-os-x), incorrect line endings could cause a similar issue. What does `cat -e /private/etc/hosts` print to the console?

ubidefeo, 2021-02-17:
closed for lack of information

closed 2021-02-17 by ubidefeo