arduino/arduino-pro-ide #383: Use of undeclared identifier with multiple .ino tabs project

novakkry, 2020-11-12:
I have a project consisting of four .ino files. The program compiles just fine but while working on it, the IDE tells me "Use of undeclared identifier" error nearly everywhere besides the first tab. Also, for instance, variable suggestions works only in the first tab, not in the others. The sketch is saved in the default user sketchbook. If I put everything into the first tab, it behives as expected.

**To Reproduce**
1. Create a multiple arduino tab project and put different components into each tab (eg. setup and loop to a second tab)

**Expected behavior**
Not showing false problems.

**Screenshots**
[image]
[image]

**Desktop (please complete the following information):**
 - OS: macOS Catalina 10.15.13
 - Version: Pro IDE 0.1.2

novakkry, 2020-11-13:
Just noticed that the red underlying problems are being showed only on the current view of the tab. If you scroll down in a tab, it's not underlined anymore, only on the first a couple of lines.

[image]

ubidefeo, 2020-11-15:
@novakkry 
A fix is in the making. We have reworked the Language Sever translator and will add the fix tot he next release.
Not sure you can disable the error highlighting, but if there is @kittaakos will have an answer for you :)

novakkry, 2020-11-15:
> @novakkry
> A fix is in the making. We have reworked the Language Sever translator and will add the fix tot he next release.
> Not sure you can disable the error highlighting, but if there is @kittaakos will have an answer for you :)

That's awesome. Thank you, guys!

ubidefeo, 2021-02-17:
resolved in `0.1.4 Alpha`

closed 2021-02-17 by ubidefeo