arduino/arduino-pro-ide #384: Editor error: 'LiquidCrystal.h' file not found. clang(pp_file_not_found)

silvioprog, 2020-11-13:
The source editor shows `'LiquidCrystal.h' file not found. clang(pp_file_not_found)` when passing the mouse above the underlined red line `23`, but the code compiles fine as shown in the picture below, so it seems a problem related to the IDE:

[image]

**Steps to reproduce the problem:**

Just install the [LiquidCrystal library](https://github.com/arduino-libraries/LiquidCrystal) and try the [`Hello world`](https://www.arduino.cc/en/Tutorial/LibraryExamples/HelloWorld) example.

**My environment (on Fedora33):**

```
Version: 0.1.2-nightly.20201112
CLI Version: 0.13.0 alpha [17d24eb]
```

ubidefeo, 2020-11-15:
hi @silvioprog 

The Language Server layer has been glitchy for a long time, we're addressing it during these weeks and plan on releasing a fix /workaround. No ETA

ubidefeo, 2021-02-17:
resolved in `0.1.4 Alpha`

closed 2021-02-17 by ubidefeo