arduino/arduino-pro-ide #385: sizeof(long) with ESP32/8266 is shown wrong

jonas-ott, 2020-11-13:
When moving the mousepointer above `sizeof(long)` with an ESP32 selected as Board the IDE shows that the value is 8. But that's wrong like shown with Serial.print. It's the same with an ESP8266 but correct with an Arduino Nano. I haven't tested other Boards.
[image]

per1234, 2021-02-17:
Hi @jonas-ott. Thanks so much for taking the time to report this!

Would you mind downloading [the new 0.1.4 release of Arduino Pro IDE](https://github.com/arduino/arduino-pro-ide/releases/tag/0.1.4) and letting us know whether the bug still occurs?

There have been some nice advancements since November and it is now showing the expected 4 value for me with both ESP32 and ESP8266, so I think it's fixed.

jonas-ott, 2021-02-17:
It didn't work for me (Manjaro Linux). So I decided to test it in a VM with Windows and there it's correct.

per1234, 2021-02-17:
Thanks for checking that for me. I was using Windows. I tried it on Linux and am getting the same results you reported on both OSes.

I notice that the Pro IDE's editor hover shows the type as `unsigned long long` on my Windows machine while it shows the type as `unsigned long` on my Linux machine (which I see is the same as in your screenshot).