arduino/arduino-pro-ide #387: IDE is not recognizing project located libraries.

MenTom-CZ, 2020-11-20:
**Describe the bug**
IDE is not recognizing project located libraries.
There is problem. When I try to incluce my library in project folder. IDE don't know about library (highlighting every usage of library in sketch(variables, functions) and tells me that everything used is undeclared) but when I try to compile my project everything isfine and compilation is succesful but ide still shows errors.

**To Reproduce**
Steps to reproduce the behavior:
1. In project folder create any .h file and add some content
2. Include that file with #include "FILENAME.h"
3. Try to acces content of file (function, variable, definition)
4. Content will bee highlighted as "Use of undeclared identifier"

**Expected behavior**
IDE should show tooltips and not highlight elements as udeclared/undefined.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - Windows 10

ubidefeo, 2020-11-21:
@MenTom-CZ 
please check out #384 
this is a known bug we're working on

ubidefeo, 2021-02-17:
resolved in `0.1.4 Alpha`

closed 2021-02-17 by ubidefeo