arduino/arduino-pro-ide #388: Library not found when compiling, but library is installed

Tonkaman, 2020-11-21:
When compiling a project using the OBD2 library, compilation fails because the IDE cannot find the .h file.  The library is on my computer, and the sketch compiles just fine in the regular Arduino IDE.  When browsing the Library Manager, the library both shows as installed and with the option to install, as if it wasn't already installed.  The version in the Library Manager shows as 0.0.0.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Windows 10 Pro
 - Version: Build 10.0.19041

**Additional context**
Add any other context about the problem here.

ubidefeo, 2020-11-22:
hi @Tonkaman 
can you copy/paste your CLI configuration file?
You can get there from the menu **File > Open CLI configuration**

Tonkaman, 2020-11-22:
board_manager:
  additional_urls: []
daemon:
  port: "50051"
directories:
  data: C:\Users\Jason\AppData\Local\Arduino15
  downloads: C:\Users\Jason\AppData\Local\Arduino15\staging
  user: C:\Users\Jason\OneDrive\Documents\Arduino
logging:
  file: ""
  format: text
  level: info
telemetry:
  addr: :9090
  enabled: true

Tonkaman, 2020-11-22:
Here is the contents of the CLI:

board_manager:
  additional_urls: []
daemon:
  port: "50051"
directories:
  data: C:\Users\Jason\AppData\Local\Arduino15
  downloads: C:\Users\Jason\AppData\Local\Arduino15\staging
  user: C:\Users\Jason\OneDrive\Documents\Arduino
logging:
  file: ""
  format: text
  level: info
telemetry:
  addr: :9090
  enabled: true

ubidefeo, 2020-11-22:
Thank you, @Tonkaman 
I trust that your library is in `C:\Users\Jason\OneDrive\Documents\Arduino\libraries`.
I have ran a few tests and this is the outcome.
I have installed the library and opened an example which seemed to be requiring `CAN.h`, which is not available through the library manager so I had to download it manually and place into `libraries` in my Sketch Folder.
I bumped into your exact issue until I tried with one of our more recent internal builds.
Please download a Nightly Build and your compilation problem should disappear
https://github.com/arduino/arduino-pro-ide#nightly-builds

Let me know if it doesn't :)

Tonkaman, 2020-11-22:
@ubidefeo 
The nightly build solves the issue.  Thanks for looking into it.  I was using the last stable release, which was dated from September.

closed 2020-11-22 by Tonkaman

ubidefeo, 2020-11-23:
@Tonkaman 
my pleasure :)
We're going to make another release soon, but trying to squeeze important fixes in it that address most complaints we've had so far.
The Nightly contains daily patches, and if you're willing to be on the edge it's the way to go ;)