arduino/arduino-pro-ide #389: Local board packages

CreativeRobotics, 2020-11-25:
In the classic Arduino IDE you can add a set of board files and a core by putting them in the hardware folder in the sketchbook - I have several that are customised versions of other distributions but which aren't packaged up and released on Github.

I can't see any way of using these boards in the Pro IDE. Have I missed something or is it a feature that doesn't exist yet?

ubidefeo, 2020-11-25:
@CreativeRobotics 
that is something that might have slipped through the cracks, although I remember talking about it.
I will investigate tomorrow with the team.
I do have some good news for you, though: we just introduced support for installing libraries from Git repository (any URL pointing to a `.git` file as well as local `.zip` files.
Download the latest Nightly Build and use the help command on `arduino-cli lib`

```shell
arduino-cli lib install -h
```
you'll see this

```
Flags:
      --git-url    Enter git url for libraries hosted on repositories
      -h, --help       help for install
      --no-deps    Do not install dependencies.
      --zip-path   Enter a path to zip file
```

so just run a `lib install` passing in `--zip-path` and you'll have installed your custom local hardware definitions :)

ubidefeo, 2020-11-26:
@CreativeRobotics I have tested with some custom hardware definitions and indeed they do not show up in the new IDE.
I've filed a bug and we'll start looking at it next week

thank you for reporting, really useful finding :)

kittaakos, 2020-11-27:
Thanks for logging it. It's already tracked [here](https://github.com/arduino/arduino-pro-ide/issues/355).

closed 2020-11-27 by kittaakos