arduino/arduino-pro-ide #390: Not enough verbosity from compiler output and possibly compiler settings in Pro IDE are wrong.

Dave528, 2020-11-27:
**Is your feature request related to a problem? Please describe.**

There is not enough detail in the compiler output to locate more than a single problem/error when compiling fails.  And the compiler is attempting to link a failed compile.  On the standard Arduino IDE there is much more detail in the error reporting from the compiler.  The project I compiled under the Pro IDE nightly from 2020-11-25 would show one ambiguous error 2 from LD,  While the same project compiled under the Arduino IDE version 1.8.13 returned dozens of specific error messages in the compiler output, allowing me to correct the (mostly) typos.  The standard IDE also had enough information to know not to bother running LD, unlike Pro IDE.

**Describe the solution you'd like**

A more verbose, clear, and concise description of the errors during compilation, rather than the errors being ignored and the IDE attempting to link broken code.  It is as though the compiler settings are causing GCC to ignore some things.

**Describe alternatives you've considered**

For now I'm using the original Arduino IDE 1.8.13 to compile my project while using the Arduino Pro for writing code.  I could not find a setting in the Pro IDE to enable more verbose output from the compiler.

**Additional context**

Sorry there are no screen-shots.  I realized after trying the Arduino IDE 1.8.13 and fixing the program that I should have recorded this.
Also, the project is a fairly large one.  And, since I'm working with an unfamiliar MCU, alot of errors in direct register addressing were introduced.  In all there were close to 100 naming errors.

ubidefeo, 2020-11-29:
hi @Dave528 
thank you for reporting this, I'll bring it to the team's attention.
can you tell me which MCU you're compiling for and with which platform core?

Dave528, 2020-11-30:
I am compiling for the Adafruit Grand Central M4.  It is a SAMD51P20A.  The processor is Arm Cortex M4F.  I'm using the Adafruit libraries for the Grand Central board (i.e. - CMSIS, with the Adafruit wrapper instead of the Arduino wrapper).

Dave528, 2020-11-30:
I should note that the project has only just begun in the last two days.  It is currently located in a single source .ino file.  The file at the time of trying to compile with the Pro IDE was approximately 1100 lines, and was simply enumerating peripheral registers from the Cache Controller (CMCC), Peripheral Access Controller (PAC), and the Timer Counter TC0 through TC7 peripherals.  The source of the problem does seem to have been overly relaxed settings as far as catching and reporting errors in code is concerned.
Thanks for your attention to this issue.
PS: The general error was in the naming of the peripheral.  I unintentionally was using the base name of the struct (for example Pac) instead of the declared name for the peripheral instance (ie. PAC) when trying to access the registers.  The compiler should have reported that I was attempting to access uninstantiated struct's.

ubidefeo, 2020-11-30:
@Dave528 
thank you for updating me on the issue.
I seem to understand you found what your problem was and am assuming you are fine with the verbosity level of the IDE,
please let me know if you still want me to look into this as an issue :)

Dave528, 2020-11-30:
I think this is an issue with options passed to the compiler.  The standard Arduino IDE is correctly having the compiler report this type of problem, as it should.  The settings passed to the compiler from the Pro IDE, however, are not.
This would seem to me to be a fairly important problem to correct.  At the very least adding settings for compiler options (which should be present anyway) would be a good idea.  After all, when projects get more complex than your typical Arduino .ino project a "one size fits all" for the compiler will be problematic.
Should I perhaps post this as a feature request?

kittaakos, 2020-12-01:
> The settings passed to the compiler from the Pro IDE, however, are not.
> This would seem to me to be a fairly important problem to correct.

The new IDE relies on the Arduino CLI when it runs the compilation. The IDE pipes the output out of the compilation into the _Output_ view. It does nothing more. If the verbosity of the compilation is not sufficient, we should move this issue to the CLI repo, should we?

Dave528, 2020-12-01:
If you think that is where this belongs.  However, if the Arduino CLI in use is the same as used by the standard IDE then it would seem the problem lies in the Pro IDE, not the CLI.  Either the Pro IDE is not correctly passing arguments to the CLI, or the Pro IDE would not be correctly listening to the CLI.  The former seems more likely.
Being unable to see the source for the Pro IDE I can only make this argument as a guess based on past experience.
So, like I said, if you think this will be better off handled as an issue with the CLI and being moved to the CLI repo then that is where it should go to.
Thanks for your time in this.  I have high hopes for the new IDE based on Theia, since it is a "real" IDE that allows much better project management.

kittaakos, 2020-12-01:
> if the Arduino CLI in use is the same as used by the standard IDE

No, it isn't.

> So, like I said, if you think this will be better off handled as an issue with the CLI and being moved to the CLI repo then that is where it should go to.

We have to get the output with the Java IDE and the CLI and compare them.

> Thanks for your time in this.

Thank you for trying out the alpha version and getting back to us with your remarks.

----

> I am compiling for the Adafruit Grand Central M4. It is a SAMD51P20A. The processor is Arm Cortex M4F. I'm using the Adafruit libraries for the Grand Central board (i.e. - CMSIS, with the Adafruit wrapper instead of the Arduino wrapper).
> (PS- Sorry for the slow reply. I've been busy today.)

@ubidefeo, could you please assist me with the steps to reproduce this 👆

Dave528, 2020-12-01:
When I compiled with the standard IDE I learned that I had unintentionally referenced the structure names (like Pac) instead of the instance names (like PAC) when accessing the peripheral registers.
After changing the names to their proper instance names the program would compile.
Example:

    i = Pac->STATUSA.reg; // Accessing via the struct from ..\Arduino\CMSIS Doc\samd51\include\component\pac.h ---- WRONG

    i = PAC->STATUSA.reg; // Accessing via the instance from ..\Arduino\CMSIS Doc\samd51\include\samd51p20a.h ---- CORRECT

I am trying this again in another new sketch, but it doesn't seem to be causing the issue.  I'm at this point unsure of how to reproduce the problem.  I'm very sorry for this.

Unfortunately I did not make a full record of what caused the compiler to not report the error like I should have.  The instinct to make it work kicked in and obfuscated the need for diligence.

The computer is:  Windows 7, 64bit, Intel i7-3632QM, 16GB RAM.  The Pro IDE was "arduino-pro-ide_nightly-20201125_Windows_64bit" (Folder in the zip file.)

closed 2020-12-01 by Dave528