arduino/arduino-pro-ide #391: Update the VSCode extension on the base of this work

Stig124, 2020-11-29:
**Is your feature request related to a problem? Please describe.**
I use VSCode Insiders as my daily-driver and I have to say something the actual Arduino extension is an absolute pain to use, the include paths aren't set up by the extension, so you'll be absolutely sure to have to drop spellcheck if you use any Libraries (even the base ones)...

**Describe the solution you'd like**
A clear and concise description of what you want to happen.
As I clearly see, this excellent work is based on VSCode, and I don't see any obvious issues that could prevent the integration of the Arduino part of the IDE into an extension that would work on any basic build of VSCode
**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.
It could also be possible to make a VSCode installer that would ship with the IDE part as a base package (basically enabling the rest of the IDE, like any build that you would download of the MS Site)
**Additional context**
Add any other context or screenshots about the feature request here.

ubidefeo, 2020-11-29:
@Stig124 
thanks a lot for your feedback and remarks.
Our new IDE is based on Theia which shares a lot with VS Code, and in fact supports extensions.
Most of the work we are doing is based on VS Code compatible extensions that we build or implement, but in order to bring it to the marketplace a lot of work is required which we don't have bandwidth for at the moment.
We keep our eyes open for opportunities, and as soon as our next milestone is reached we'll move the roadmap forward and explore our opportunities for the future :)

Stig124, 2020-11-30:
Thanks for your fast response !

closed 2020-11-30 by Stig124