arduino/arduino-pro-ide #392: Notification: Error during build: exec: "g++": executable file not found in %PATH%

JSCLadon, 2020-12-02:
**Describe the bug**
A Notification says: "Could not start editor support. Error during build: exec: "g++": executable file not found in %PATH%"
I get a clean compile and the sketch works as expeted. But the notification is really anoying.

**To Reproduce**
I have it all the time, no matter what sketch I am using

**Expected behavior**
Go away ;-)

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Version: 
@theia/callhierarchy 1.5.0-next.09ef844b
@theia/console 1.5.0-next.09ef844b
@theia/core 1.5.0-next.09ef844b
@theia/debug 1.5.0-next.09ef844b
@theia/file-search 1.5.0-next.09ef844b
@theia/filesystem 1.5.0-next.09ef844b
@theia/git 1.5.0-next.09ef844b
@theia/languages 1.5.0-next.09ef844b
@theia/markers 1.5.0-next.09ef844b
@theia/outline-view 1.5.0-next.09ef844b
@theia/output 1.5.0-next.09ef844b
@theia/plugin-ext 1.5.0-next.09ef844b
@theia/plugin-ext-vscode 1.5.0-next.09ef844b
@theia/process 1.5.0-next.09ef844b
@theia/scm 1.5.0-next.09ef844b
@theia/scm-extra 1.5.0-next.09ef844b
@theia/search-in-workspace 1.5.0-next.09ef844b
@theia/task 1.5.0-next.09ef844b
@theia/terminal 1.5.0-next.09ef844b
@theia/userstorage 1.5.0-next.09ef844b
@theia/variable-resolver 1.5.0-next.09ef844b
@theia/workspace 1.5.0-next.09ef844b
arduino-debugger-extension 0.1.0
arduino-ide-extension 0.1.0
arduino-cli.exe Version: 0.12.0 Commit: d68544e

**Additional context**
Had some problems to find the lib earlier, but those are gone, since I am loading the lib from the scetch folder.

kittaakos, 2020-12-15:
Thanks for logging, @JSCLadon.

> arduino-ide-extension 0.1.0
> arduino-cli.exe Version: 0.12.0 Commit: d68544e

Your version is quite old, could you please get the [nightly](https://github.com/arduino/arduino-pro-ide#nightly-builds) and try again?

JSCLadon, 2020-12-28:
Problem solved with the newer version. Thanks a lot! Really appeciate your great work!

closed 2020-12-28 by JSCLadon