arduino/arduino-pro-ide #393: Language server not started

peter76, 2020-12-11:
**Describe the bug**
When I start Arduino pro ide 0.1.3 and select nano as board I get the following error:

"Couldn't start client arduino language server"

This is some relevant output in the terminal when this happens:

`root INFO Board config changed: 
root INFO Starting language server: arduino:avr:nano
root ERROR [hosted-plugin: 6465] Promise rejection not handled in one second: Error: json: cannot unmarshal bool into Go struct field ServerCapabilities.capabilities.textDocumentSync of type lsp.SaveOptions , reason: Error: json: cannot unmarshal bool into Go struct field ServerCapabilities.capabilities.textDocumentSync of type lsp.SaveOptions
root ERROR [hosted-plugin: 6465] With stack trace: Error: json: cannot unmarshal bool into Go struct field ServerCapabilities.capabilities.textDocumentSync of type lsp.SaveOptions
    at handleResponse (/home/peter/Apps/arduino-pro-ide_0.1.3_Linux_64bit/resources/app/plugins/vscode-arduino-language-server/extension/dist/bundle.js:4725:48)
    at processMessageQueue (/home/peter/Apps/arduino-pro-ide_0.1.3_Linux_64bit/resources/app/plugins/vscode-arduino-language-server/extension/dist/bundle.js:4552:17)
    at Immediate.<anonymous> (/home/peter/Apps/arduino-pro-ide_0.1.3_Linux_64bit/resources/app/plugins/vscode-arduino-language-server/extension/dist/bundle.js:4536:13)
    at processImmediate (internal/timers.js:439:21)
`
Let me know if you need more info.

**Desktop (please complete the following information):**
 - OS: Fedora
 - Version: 33

Regards, Peter

kittaakos, 2020-12-14:
@peter76, how did you start debugging? Debugging should be disabled from the toolbar when you select the Nano board. It works with boards that support debugging: Zero, MKR1000.

[image]

Did you start debugging manually from the _Debug_ view?

peter76, 2020-12-14:
@kittaakos , I just checked, but I didn't start debuggging. When I now start Arduino-pro-ide I get an empty sketch, with no board selected and debugging greyed out in the toolbar, like in your screesnshot.
When I select the nano, or any other board, I get the "Couldn't start client arduino language server" message.
But I guess from the command line output you see that debugging is enabled?
Thanks for looking into this, the ide looks very promising:-)

Regards, Peter

kittaakos, 2020-12-14:
> I just checked, but I didn't start debuggging.

👍 

> When I select the nano, or any other board, I get the "Couldn't start client arduino language server" message.

OK. I need to set up a Linux VM and try it out. Thank you for your persistence on this bug and convincing me it's not a debugger issue.

peter76, 2020-12-14:
Ok, thanks, let me know if I can do anything on my side

McWeird, 2020-12-14:
In case it is of any help I'm also running into this with 0.1.3 on Ubuntu 20.04, but it appears to be only in the cases where the sketches include libraries. Let me know if i can be of any help.

peter76, 2020-12-14:
That's strange... I just created a new user account and downloaded 0.1.3 and ran the ide in this new account,
but exactly the same, just with an empty sketch...

kittaakos, 2020-12-15:
@peter76, @McWeird how did you unpack the IDE? Maybe you have the same issue as described [here](https://github.com/arduino/arduino-pro-ide/issues/213#issue-571281519). Could you please run:
```
/path/to/arduino-pro-ide_0.1.3_Linux_64bit/resources/app/node_modules/arduino-ide-extension/build/bin/clangd --version
```

peter76, 2020-12-15:
@kittaakos, I unpacked the ide with Fedoras standard unarchiver. If I run your command above I get:

`clangd version 9.0.0 (tags/RELEASE_900/final)`

So that seems to be ok. I also checked if the symbolic links are in place from the other bug you mention, here is the out put:

`lrwxrwxrwx. 1 peter peter      14 11 dec 20:44 libc++abi.so -> libc++abi.so.1
lrwxrwxrwx. 1 peter peter      16 11 dec 20:44 libc++abi.so.1 -> libc++abi.so.1.0
-rw-r--r--. 1 peter peter  308288 22 sep  2019 libc++abi.so.1.0
-rw-r--r--. 1 peter peter      28 11 dec 14:15 libc++.so
lrwxrwxrwx. 1 peter peter      13 11 dec 20:44 libc++.so.1 -> libc++.so.1.0
-rw-r--r--. 1 peter peter 1017080 22 sep  2019 libc++.so.1.0
lrwxrwxrwx. 1 peter peter      15 11 dec 20:44 libtinfo.so.5 -> libtinfo.so.5.9
-rwxr-xr-x. 1 peter peter  193928 21 jan  2020 libtinfo.so.5.9`

I have Visual Studio Code installed system wide, could there be a conflict with that?

peter76, 2020-12-15:
@kittaakos , I just redownloaded the ide and unpacked it through the command-line with `unzip` , but get the same error about the language server.

kittaakos, 2020-12-15:
OK, thanks for checking. We found a Linux machine that fails with the same issue; we're on it.

kittaakos, 2020-12-15:
OK, it seems there is an issue if you have `clangd` in the `PATH`. At least on our side, we could consistently reproduce the defect. The Arduino LS does not start if,
 - `clangd` (`11.x`) is available from the `PATH`, the LS does not start.
 - `clangd` is **not** in the `PATH`, it works.

peter76, 2020-12-15:
Does that mean that the ide should prefer it's bundled clang version instead of the system wide installed version?

Regards, Peter

McWeird, 2020-12-15:
I unpacked through drag and drop, the output of your command is the same as peter's with:

```clangd version 9.0.0 (tags/RELEASE_900/final)```

Unpacking through unzip still gives the same issue. Output of `ls -l resources/app/node_modules/arduino-ide-extension/build/lib/` is:

```
total 1496
lrwxrwxrwx 1 ferran ferran      14 de des.  15 10:31 libc++abi.so -> libc++abi.so.1
lrwxrwxrwx 1 ferran ferran      16 de des.  15 10:31 libc++abi.so.1 -> libc++abi.so.1.0
-rw-r--r-- 1 ferran ferran  308288 de set.  22  2019 libc++abi.so.1.0
-rw-r--r-- 1 ferran ferran      28 de des.  11 14:15 libc++.so
lrwxrwxrwx 1 ferran ferran      13 de des.  15 10:31 libc++.so.1 -> libc++.so.1.0
-rw-r--r-- 1 ferran ferran 1017080 de set.  22  2019 libc++.so.1.0
lrwxrwxrwx 1 ferran ferran      15 de des.  15 10:31 libtinfo.so.5 -> libtinfo.so.5.9
-rwxr-xr-x 1 ferran ferran  193928 de gen.  21  2020 libtinfo.so.5.9

kittaakos, 2020-12-15:
I could reproduce the issue by installing clangd 11.x into my macOS (`brew install llvm`) and starting the new Arduino IDE:
```
root INFO Starting language server: arduino:avr:uno
root ERROR [hosted-plugin: 6586] Promise rejection not handled in one second: Error: json: cannot unmarshal bool into Go struct field ServerCapabilities.capabilities.textDocumentSync of type lsp.SaveOptions , reason: Error: json: cannot unmarshal bool into Go struct field ServerCapabilities.capabilities.textDocumentSync of type lsp.SaveOptions
root ERROR [hosted-plugin: 6586] With stack trace: Error: json: cannot unmarshal bool into Go struct field ServerCapabilities.capabilities.textDocumentSync of type lsp.SaveOptions
    at handleResponse (/Users/akos.kitta/git/arduino-editor/plugins/vscode-arduino-language-server/extension/dist/bundle.js:4725:48)
    at processMessageQueue (/Users/akos.kitta/git/arduino-editor/plugins/vscode-arduino-language-server/extension/dist/bundle.js:4552:17)
    at Immediate.<anonymous> (/Users/akos.kitta/git/arduino-editor/plugins/vscode-arduino-language-server/extension/dist/bundle.js:4536:13)
    at processImmediate (internal/timers.js:439:21)
```

> Does that mean that the ide should prefer it's bundled clang version instead of the system wide installed version?

Yes. The IDE is configured to use it's embedded `clangd`. Obviously, there is a bug somewhere.

Thank you all for the help 🙏 , I am trying to come up with a fix today and roll-out a new version.

McWeird, 2020-12-15:
Good luck and thanks for looking into it! Will keep track of this.

closed 2020-12-15 by kittaakos

kittaakos, 2020-12-15:
I have fixed the issue and re-published the artifacts, could you please pull one of the [nightlies](https://github.com/arduino/arduino-pro-ide#nightly-builds) and try it again. If it still does not work, I will reopen the issue. Please note, the fix does not fix all the issues we have general with the Arduino language, but the underlying `clangd` process musty not crash. We are still working on the language feature. Thank you!

peter76, 2020-12-15:
@kittaakos Works here! FYI I unzipped it with fedora's unarchiver, so that doesn't seem to give problems on Fedora. Thanks and keep up the good work and happy holidays!

McWeird, 2020-12-15:
Same here, back to tinkering on the board, and thanks for the hard work!

kittaakos, 2020-12-15:
Thank you for your help, @peter76, and @McWeird