arduino/arduino-pro-ide #394: Serial Console Update Rate issue.

cl330b, 2020-12-12:
The Serial Monitor built in to the IDE has a significantly delayed update rate.

I am using the Arduino Zero. I am using the following to print serial information to the internal Serial Monitor and an external bit of software (GTKTerm v1.1) to monitor the serial information being written to the native USB port. 

` Serial.begin(115200);
  SerialUSB.begin(115200);`

1. Open built in Serial Monitor.
2. Open external Serial Monitor.
3. Open scope to view which output is currently active.
4. View lag in the built in Serial Monitor.
5. Compare the lag with an external Serial Monitor.

I would expect to see the update rate of the built in Serial Monitor match the modulated outputs of the microcontroller. I am using a scope to view the outputs of the Zero. I can see in real time which outputs are on compared to the built in Serial Monitor and the external Serial Monitoring software.

(Please see attached screenshots for context.)
[image]
[image]
[image]

**Version Information:**
`Linux 5.8.0-31-generic #33-Ubuntu SMP Mon Nov 23 18:44:54 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux` 
`arduino-pro-ide_nightly-20201211_Linux_64bit`

**Additional Information Regarding Issue**
When I hold the board in reset, the built in Serial Monitor continues to update as if still reading incoming serial data.

-cl330b

cl330b, 2020-12-12:
As I continue to test, I discovered that I am not seeing the issue when I serial print `millis()` to both the internal Serial Monitor and an external Serial Monitor. I will make some code modifications and if this issue is self-inflicted i will close the issue.

cl330b, 2020-12-13:
When I slow the code down, the Serial Monitor can keep up. In the above screenshots I was writing a serial message every 1.37mS (0.00137sec). I have slowed the code down to write to serial every 1000mS (1sec). The time it takes to print the characters on the built in Serial Monitor takes a bit longer than one full second to print roughly 24 characters (including spaces).

**Screenshots**
[image]
[image]
[image]

**Note**
_Please let me know if there is something specific I can dig into to provide more information._

kittaakos, 2020-12-14:
> Please let me know if there is something specific I can dig into to provide more information.

Well, I doubt this issue belongs here but since you have taken the time to precisely submit the details, I leave it open. Perhaps someone with better domain knowledge can decide if this is something we have to work on or not.

CC: @ubidefeo

ubidefeo, 2020-12-29:
hi @cl330b 

thank you so much for going through the effort of connecting your DSO, tells us you've done more than we expect when it comes to feedback. 💯 

We have received another report of the serial monitor not able to keep up #400 and this is definitely something that needs work, especially as we work towards faster devices having to provide serial monitor output.

We'll keep this one open and provide updates in the weeks to come, but we have quite a list of things that need to be worked on before we tackle the Serial Monitor

Hope to have updates soon
u.