arduino/arduino-pro-ide #395: No full screen on my 15" laptop

Attigliuzzo, 2020-12-12:
I use the latest Arduino pro ide pro version on my pc with a 27" monitor and on my lenovo g580 (15" display, 1366x768) both equipped with  Debian testing. I get full screen on my pc but this doesn't happen on my laptop, i'm unable to set the full screen app window.
I think this app has a default minimal dimension, but is possible to edit it?
My screen: (https://i.postimg.cc/T2Hfj26K/photo5956281979651601106.jpg)
I tried to stretch the window but i lose the bottom of the window.

kittaakos, 2020-12-14:
Thanks for letting us know. It's a known issue. For the sake of simplicity, I would like to avoid duplicating issues. Please subscribe to https://github.com/arduino/arduino-pro-ide/issues/351.

closed 2020-12-14 by kittaakos

Attigliuzzo, 2020-12-14:
I'm so sorry, i should have searched that issue before open new one.

kittaakos, 2020-12-14:
> I'm so sorry

No need to. It's better to log things twice than forgetting about them. 👍