arduino/arduino-pro-ide #396: Could not dispose connection. & Error: Connection got disposed.

cl330b, 2020-12-13:
**Problem Description**
While attempting to upload a sketch I am currently working on, I press the "Upload" button and nothing happens. After five to ten seconds, the Output displays `cortex_m reset_config sysresetreq`, then two errors `Error: CMSIS-DAP command CMD_INFO failed.` and `Error: No Valid JTAG Interface Configured.`. Finally it exits with status 255.

**Console Logfile**
Attached is the Console logfile.
[-1607824743807.log](https://github.com/arduino/arduino-pro-ide/files/5684089/-1607824743807.log)

The ports have not changed, the board has not been powered off, the PC has not been restarted. Nothing has changed from previously being able to upload a sketch without issue no more than a few minutes prior.

**To Reproduce**
1. Modify existing sketch
2. Attempt to upload to board.

**Expected behavior**
I was expecting the sketch to load.

**Screenshots**
Here is what the "Output" shows when this issue occurs.

[image]

**Version Information:**
`Linux 5.8.0-31-generic #33-Ubuntu SMP Mon Nov 23 18:44:54 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux`
`arduino-pro-ide_nightly-20201211_Linux_64bit`

**Additional Details:**
When I press the "Upload" again, the Console displays:

`reconnecting-websocket-mjs.js:180 
ErrorEvent {target: ReconnectingWebSocket, type: "error", message: "TIMEOUT", error: Error: TIMEOUT
    at ReconnectingWebSocket../node_modules/reconnecting-websocket/dist/reconnecting…}`

cl330b, 2020-12-13:
I am using Arduino Zero for these tests. For the above issue, it appears as if the port assignments switched. Prior to this issue, the programming port of the Zero was on `/dev/ttyACM0` and the native USB port was on `/dev/ttyACM1`. In the screenshot below, the board/port selection drop down now shows the Zero's programming port appears to be on `/dev/ttyACM1`. 

I have selected the programming port on `/dev/ttyACM1` and attempted to upload to the Zero. Still receive the same two errors as above. And then finally the "Upload Error" with exit status 255.

**Errors Received:**
1. `Error: CMSIS-DAP command CMD_INFO failed`
2. `Error: No Valid JTAG Interface Configured`
3. `Upload error: Error: 2 UNKNOWN: uploading error: uploading error: exit status 255`

**Board/Port Selection Dropdown Screenshot:**

[image]

**Resolution:**
I disconnect both the programming port and the native USB port from the Zero. Then plug back in. I select the programming port on `/dev/ttyACM0` and upload the sketch. 

**Thoughts:**
I will pay close attention to exactly what takes place prior to this issue presenting itself.

kittaakos, 2020-12-14:
Checking the logs you have attached, it seems to me there is an issue with the upload while you're connected to the serial monitor. The upload does not work because the serial monitor is using the same port. Can you please try to disconnect the monitor, and try upload again?

kittaakos, 2020-12-14:
> pload while you're connected to the serial monitor

Maybe it's due to the external monitor you have attached to the same port: https://github.com/arduino/arduino-pro-ide/issues/394

Of course, the IDE cannot ask other programs to disconnect from the port you want to use.

cl330b, 2020-12-16:
I am a bit embarrassed. Yes I am aware that I need to disconnect the Serial monitors before uploading. For some reason the upload worked many times under this condition. GTKTerm would display the port as closed while uploading almost every time. I figured I was good to go....

I have disabled serial all together and have not had the issue since. I will close this issue.

closed 2020-12-16 by cl330b

kittaakos, 2020-12-16:
> I have disabled serial all together and have not had the issue since. I will close this issue.

Thank you so much for getting back to us and updating the issue. 🙏