arduino/arduino-pro-ide #397: No port discovered

Do4kgt, 2020-12-15:
Hello everybody.

I had a total crash since yesterday. I had the mistake
13 Internal: Received RST_Stream with code 2 (Internal Server Error)

I then installed Git on Windows 10 and since then I can compile again.

however, since yesterday I can no longer select a board although it is recognized by the system.

I just can't choose a port and don't know what to do anymore.

[image]

Do4kgt, 2020-12-15:
noport

kittaakos, 2020-12-16:
@Do4kgt, thanks for reporting.

 - Attach your board,
 - Open a terminal (e.g, `CMD.EXE`),
 - Run `path\\to\\the\\arduino\\IDE\\resources\\app\\node_modules\\arduino-ide-extension\\build\\arduino-cli.exe board list --format json`.

What is the output?

When I attach my Uno board, I see this:
```bash
C:\>Users\\kittaakos\\Desktop\\arduino-pro-ide_0.1.3-snapshot.08291f5_Windows_64bit\\resources\\app\\node_modules\\arduino-ide-extension\\build\\arduino-cli.exe board list --format json
[
  {
    "address": "COM3",
    "protocol": "serial",
    "protocol_label": "Serial Port (USB)"
  },
  {
    "address": "COM4",
    "protocol": "serial",
    "protocol_label": "Serial Port (USB)"
  },
  {
    "address": "COM7",
    "protocol": "serial",
    "protocol_label": "Serial Port (USB)",
    "boards": [
      {
        "name": "Arduino Uno",
        "FQBN": "arduino:avr:uno",
        "VID": "0x2341",
        "PID": "0x0043"
      }
    ]
  }
]

C:\>
```

ubidefeo, 2021-02-17:
Closed for lack of information 2 months later.

closed 2021-02-17 by ubidefeo