arduino/arduino-pro-ide #399: Auto Format

NoSmoke999, 2020-12-17:
**Describe the bug**
Auto Format (ctrl-T) doesn't seem to do anything (unlike the non-pro IDE).

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on 'Tools/Auto Format'
3. Scroll down to '....'
4. See error

**Expected behavior**
Code to be formatted.

**Desktop (please complete the following information):**
 - OS: [Win 10 Home 64 bit]
 - Version: [Version	10.0.19041 Build 19041]

**Additional context**
Add any other context about the problem here.

ubidefeo, 2020-12-18:
@kittaakos can you look into this one?

ubidefeo, 2020-12-27:
@NoSmoke999 
we have been heavily reworking the Language Server feature and introduced this bug.
We'll address it when the team is back from the holidays :)

NoSmoke999, 2020-12-28:
OK, thanks.  :)