arduino/arduino-pro-ide #400: Ultrasonic module code works in classic IDE but not in Pro.

NoSmoke999, 2020-12-17:
**Describe the bug**

The following program uses the HC-SR04 ultrasonic module to measure distance of an object from the module.

It works as expected with Arduino IDE but gives erratic measurement results with Pro.  It will in fact produce measurement values whether or not the module is wired to the Arduino Uno.  Faulty compilation of pulseIn()?

**To Reproduce**

// ---------------------------------------------------------------- //
// Arduino Ultrasoninc Sensor HC-SR04
// Re-writed by Arbi Abdul Jabbaar
// Using Arduino IDE 1.8.7
// Using HC-SR04 Module
// Tested on 17 September 2019
// ---------------------------------------------------------------- //

#define echoPin 2 // attach pin D2 Arduino to pin Echo of HC-SR04
#define trigPin 3 //attach pin D3 Arduino to pin Trig of HC-SR04

// defines variables
long duration; // variable for the duration of sound wave travel
int distance; // variable for the distance measurement

void setup() {
  pinMode(trigPin, OUTPUT); // Sets the trigPin as an OUTPUT
  pinMode(echoPin, INPUT); // Sets the echoPin as an INPUT
  Serial.begin(9600); // // Serial Communication is starting with 9600 of baudrate speed
  Serial.println("Ultrasonic Sensor HC-SR04 Test"); // print some text in Serial Monitor
  Serial.println("with Arduino UNO R3");
}

void loop() {
  // Clears the trigPin condition
  digitalWrite(trigPin, LOW);
  delayMicroseconds(2);
  // Sets the trigPin HIGH (ACTIVE) for 10 microseconds
  digitalWrite(trigPin, HIGH);
  delayMicroseconds(10);
  digitalWrite(trigPin, LOW);
  // Reads the echoPin, returns the sound wave travel time in microseconds
  duration = pulseIn(echoPin, HIGH);
  // Calculating the distance
  distance = duration * 0.034 / 2; // Speed of sound wave divided by 2 (go and back)
  // Displays the distance on the Serial Monitor
  Serial.print("Distance: ");
  Serial.print(distance);
  Serial.println(" cm");
}

**Expected behavior**
Proper distance measurement results from the Pro version.

**Desktop (please complete the following information):**
 - OS: [Win 10]
 - Version: [Pro IDE 0.1.3]

ubidefeo, 2020-12-18:
@NoSmoke999 
the fact that you get output even though the sensor is not plugged in is due to `pulseIn()` doing its job of reading the pin and calculating the width of the pulse.
The sensor is not smart and there's no communication protocol or handshaking, it's just code and microcontroller trickery.
Compilation with both IDEs should produce the same results.

can you copy/paste the results from the Java IDE and the Pro IDE?
might just be the way text is printed in the serial monitor by the latter

u.

NoSmoke999, 2020-12-18:
Hi and thank you for the quick response.  I have performed some more testing and think I may have found the issue.  The problem appears to be in the serial monitor - the Pro IDE monitor seems unable to keep up with the rapid output from the program and thus gets way behind whereas the Java IDE prints much faster and is able to keep up.  This puts the Pro output seriously out of sync with changes made to the distance of the ultrasound reflecting surface and thus the appearance of erratic operation.   If however I put a 500 ms delay in the program to slow down the serial output, both IDEs seem to work properly.

I hope that sheds some light.  I will let you know if I find anything else.

BTW, I have ordered a SAMD device (Arduino MO Pro) and look forward especially to trying out the debug capability!
Thanks for all your efforts in this project.

ubidefeo, 2020-12-27:
hey @NoSmoke999 

latency in Serial Throughput on the host side can be an annoyance.
thank you for reporting this, we'll look into it :)

NoSmoke999, 2020-12-28:
Thank you.  :)

BTW, just discovered that Trusteer (a security program) can drastically slow down compiles.  Remove it and all is well.

ubidefeo, 2020-12-28:
hey @NoSmoke999 
I'm not sure there's a debug recipe for the M0 Pro in the platform, since it's a retired product, but let me know if you can't go ahead and I'll try and help you out.

>Thanks for all your efforts in this project.

we really appreciate this :)

NoSmoke999, 2020-12-28:
Thank you for that.  I have not received the MO pro yet but will try it out as soon as it arrives and let you know if any issues arise.
:)

per1234, 2021-02-19:
Hi @NoSmoke999 . Thanks for your valuable issue report!

I see you were able to determine the cause of the problem is the limited Serial Monitor update rate. We already have a prior issue report about this at https://github.com/arduino/arduino-pro-ide/issues/394.

It's best to have only a single issue per subject so we can consolidate all relevant discussion to one place, so I'll go ahead and close this in favor of that one. If you end up with additional information to share, feel free to comment in the other thread.

closed 2021-02-19 by per1234