arduino/arduino-pro-ide #401: Debian Buster execution

jmarkuna, 2020-12-18:
root ERROR Uncaught Exception: 
root ERROR Error: ENOENT: no such file or directory, stat 'undefined/'
    at statSync (fs.js:915:3)
    at Object.fs.statSync (electron/js2c/asar.js:278:27)
    at mkDir (/spare/downloads/arduinoProIDE/arduino-pro-ide_0.1.3_Linux_64bit/resources/app/node_modules/node-log-rotate/lib/utils/find-log-path.js:52:17)
    at prepareDir (/spare/downloads/arduinoProIDE/arduino-pro-ide_0.1.3_Linux_64bit/resources/app/node_modules/node-log-rotate/lib/utils/find-log-path.js:31:9)
    at Object.default_1 [as default] (/spare/downloads/arduinoProIDE/arduino-pro-ide_0.1.3_Linux_64bit/resources/app/node_modules/node-log-rotate/lib/utils/find-log-path.js:8:19)
    at Object.default_1 [as default] (/spare/downloads/arduinoProIDE/arduino-pro-ide_0.1.3_Linux_64bit/resources/app/node_modules/node-log-rotate/lib/utils/find-log-file-name.js:5:36)
    at Object.default_1 [as default] (/spare/downloads/arduinoProIDE/arduino-pro-ide_0.1.3_Linux_64bit/resources/app/node_modules/node-log-rotate/lib/transport-file.js:15:52)
    at log (/spare/downloads/arduinoProIDE/arduino-pro-ide_0.1.3_Linux_64bit/resources/app/node_modules/node-log-rotate/lib/index.js:20:36)
    at Object.<anonymous> (/spare/downloads/arduinoProIDE/arduino-pro-ide_0.1.3_Linux_64bit/resources/app/src-gen/backend/main.js:11:9)
    at Object.log (/spare/downloads/arduinoProIDE/arduino-pro-ide_0.1.3_Linux_64bit/resources/app/node_modules/@theia/core/lib/common/logger-protocol.js:141:17)

ran arduino-pro-ide. funny thing is I cannot strace arduino-pro-ide system says cannot stat file.
Arduino pro gui comes up but I keep getting errors shown about. I can be root on this system, have tried both root acct and non-root account to run arduino-pro-ide

ubidefeo, 2020-12-29:
hi @jmarkuna 

which machine are you running this on?

jmarkuna, 2020-12-29:
I am running this on a Linux Debian buster OS latest revision a Dell
Optiplex model

ubidefeo, 2021-02-17:
@jmarkuna 
any chance you have tried a more recent release of this IDE on your installation?