arduino/arduino-pro-ide #403: Dark theme how to set.

djmcg, 2020-12-20:
Hello
I`m try to change default theme for dark.
Could You advice how to do it.?>

djmcg, 2020-12-20:
Finally I find it:
`Ctrl+Shift+P
[image]
`

ubidefeo, 2020-12-27:
hi @djmcg 
the theme setting belongs to Theia and being still in alpha we don't have many functionalities exposed yet.
happy you found it, consider this an Easter Egg... or a 🎄 one? 😬

NoSmoke999, 2020-12-28:
Ctrl+Shift+P gives a pop-up for all the keyboard shortcuts on my version (0.1.3).  Has there been a new release?

ubidefeo, 2020-12-28:
@NoSmoke999 
correct, with that shortcut you get access to a lot of Theia configuration stuff.
When I say "exposed" I mean through a more user-friendly UI
0.1.3 is the latest

sebasdt, 2021-02-06:
> Ctrl+Shift+P gives a pop-up for all the keyboard shortcuts on my version (0.1.3). Has there been a new release?

look in preferences or CTRL+Komma. and theme

ubidefeo, 2021-02-07:
@sebasdt the Nightly Build has Themes selectable from the Preferences GUI ;)