arduino/arduino-pro-ide #405: Distribute features as VS Code extension.

Marsfan, 2021-01-01:
## **Is your feature request related to a problem? Please describe.**
The new Arduino Pro IDE looks really nice and it is a massive improvement over the current IDE, but it is becoming a royal pain to have to install a new IDE for every different board/framework that I work with (Arduino, MBED OS, Cypress PSOC, Microchip PIC, etc). 

## **Describe the solution you'd like**
Release the Pro IDE not only as a standalone application, but as an extension that users can download into Visual Studio Code. 

## **Describe alternatives you've considered**
All of the alternatives that I have listed here do indeed work, but none of them would be as simple as just installing an extension through visual studio code. 

### [Arduino Pro Ide](../)
* **Advantages**
  * It just works
  * Similar UI to existing Arduino IDE
* **Disadvantages**
  * Requires installer a whole new IDE just to program Arduino
  * No Additional Board manager URL in GUI Config
### [Microsoft Arduino Extension](https://github.com/Microsoft/vscode-arduino)
* **Advantages**
  * Quick and easy to install in VS Code
  * Works out of the box if Arduino IDE is already installed
* **Disadvantages**
  * UI is of low quality
  * Lacks ability to identify problems without verifying/compiling sketch
  * Output from verification/compilation is messy, due to use of `arduino_debug.exe`
  * Output from verification/compilation does not appear in problems list, nor displays in color. 
  * No Additional Board manager URL Config Menu
### [VS Code Task creation](https://code.visualstudio.com/docs/editor/tasks)
* **Advantages**
  * Lots of Customization 
  * Built into VS Code
  * Supports Problem Matching with VS Code
* **Disadvantages**
  * Requires lots of manual tweaking for each individual project.
  * Cannot easily define a GUI-based method for defining board/serial port.
  * No Boards or Library Manager, must use Arduino IDE to add new boards or libraries. 
### [Platformio IDE](https://platformio.org/)
* **Advantages**
  * Large Feature Set (Even more features than Pro IDE depending on board)
  * Available as an extension for many different IDE
  * Features some board not currently supported in the Arduino IDE
* **Disadvantages**
  * Does not support all Arduino Compatible board (Sparkfun Artemis being an example at the time of writing)
  * Requires additional configuration to support using Arduino Framework. 

## **Additional context**
* The release notes in the Pro IDE mention moving the language feature to be a VS Code extension. This suggests that the framework to bundle the features into is already in place behind the scenes (which I cannot verify as the project is not open-source). Furthermore, as VS Code supports extension bundles (see [vscode-pack-arduino)[https://github.com/mpty-ltd/vscode-pack-arduino/] as an example), refactoring existing extensions into one full extension is not necessary. 
* VS Code supports adding a preview flag, to indicate to users that the extension is still in beta and subject to change. (See [Microsoft's own PowerShell extension as an example](https://github.com/PowerShell/vscode-powershell), specifically [Line 5 of package.json](https://github.com/PowerShell/vscode-powershell/blob/308b7ce7bc9f8f44271718f674d91a2ecab03f8a/package.json#L5)
* While this suggestion adds complication to the project, moving forward with open-sourcing this project (as discussed in the early announcement blog post), can mitigate this by adding many more developers to the project. I am sure that most developers interested in contributing to this project will not mind having to work around with alpha stage code.

JSCLadon, 2021-01-05:
I would strongly support Marsfan's enhancement request! I am working with the Portenta and I am using the Arduino IDE PRO only for the portenta. Although the PRO is a big step forward, it still is lacking some features, which are already available f. e.g. with VS. Some providers like Visual Micro claim to have a plugin for Arduino, but I [never got it to work sufficiently](https://www.visualmicro.com/forums/YaBB.pl?num=1609688781).

ubidefeo, 2021-01-05:
@Marsfan @JSCLadon 

at this moment we, as a very small team, are focusing on getting this IDE ready for prime-time, and the decision to use and develop VS Code compatible extensions makes our work easier.
Most of the components you see and use in the new IDE have some features which have been developed to work with the underlying Theia IDE, and I'm sure they'd work on VS Code with some modifications, but until we have this product ready we will not be looking at that option.

VS Code is an incredible IDE, our whole team agrees, and everyone uses it.
We'll see where we move towards in the future, our work here is not done yet :D

Marsfan, 2021-01-05:
@ubidefeo  Makes total sense. Any clue on when you are going to release the source code? I would be happy to help get it more feature-complete if the source code was available

ubidefeo, 2021-01-05:
@Marsfan 
we are going to release once the codebase is clean and we have some documentation.
In full transparency we can't wait for this to go OS, but we have to do it right :)