arduino/arduino-pro-ide #406: No code completion on Mac

jtmaston, 2021-01-03:
Hello! On a windows installation, I used to have autocompletion ( for things like `Serial` or the many variable types ) however, since I moved to Mac this feature is no longer available. Autocompletion does work with variables I have declared myself, however. I would like to note that I am on an older version of MacOS ( 10.11.6 ), so maybe that's the issue? Any help would be greatly appreciated.

ubidefeo, 2021-01-03:
@jtmaston 
the auto-complete functionality is currently broken in Alpha
we're working towards a solution but it's proving to be a stubborn one

within a couple of weeks we should have an update regarding this issue,
but right now I can't estimate when it will be resolved

thank you
Ubi

jtmaston, 2021-01-07:
Thank you for your input! Waiting enthusiastically!

closed 2021-01-07 by jtmaston