arduino/arduino-pro-ide #409: macOS, cannot open a file directly from Finder

sstaub, 2021-01-11:
**Describe the bug**
On macOS it is not possible to open a file directly from the Finder.
This possible with the classic Arduino IDE.
On double click or right click / open with... starts the Pro IDE but not with right file

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Finder
2. Open a folder with an .ino file inside
3. Double click for opening
4. IDE opens but not with the file you clicked

**Expected behavior**
It should possible to open an .ino file directly from the Finder, like on Classic IDE.
Also with a message when there is no right folder -> auto create a folder

**Desktop (please complete the following information):**
 - OS: macOS 10.15.7
 - Version: Alpha 0.1.3

ubidefeo, 2021-01-11:
@sstaub 
We are aware of this.
The behaviour your describe is default on the Java IDE, 
but we have yet to implement it here.

We'll leave the issue open for the time being, thank you for reporting it 👍🏼

sstaub, 2021-01-12:
@ubidefeo 
I made a patch for the Nucleo-144 boards. I will attach it here, the paths are original.
Following PinNames.h changed:
Nucleo-F207ZG
Nucleo-F303ZE
Nucleo-F412ZG
Nucleo-F413ZH
Nucleo-F429ZI
Nucleo-F446ZE
Nucleo-F746ZG
Nucleo-F756ZG
Nucleo-F767ZI
Nucleo-H7A3ZI-Q
Nucleo-H743ZI
Nucleo-H743ZI2

[TARGET_STM.zip](https://github.com/arduino/arduino-pro-ide/files/5801791/TARGET_STM.zip)