arduino/arduino-pro-ide #411: Debug fails with Arduino M0

NoSmoke999, 2021-01-19:
Debug will not initiate properly with Arduino M0 (Native USB Port).  When "Start Debugging" is tried, the following error message results:

Open On-Chip Debugger 0.10.0+dev-gf0767a31 (2018-06-11-13:36)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "swd". To override use 'transport select <transport>'.
none separate
adapter speed: 400 kHz
cortex_m reset_config sysresetreq
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Error: unable to find CMSIS-DAP device

 Win 10 Home, latest Arduino-Pro IDE release

Thanks for any assistance :)

per1234, 2021-01-19:
Hi @NoSmoke999. Do you have a CMSIS-DAP debug probe connected to your computer?

NoSmoke999, 2021-01-19:
Hi per1234.  No, I did not know one was needed (I'm a newbie with this kind of stuff).  I kinda thought the SAMD boards, including the M0, came with the required circuitry.

I see the CMSIS probes can be purchased online and they appear to connect to a SMD port (which the M0 has - a 2x5 pin connector).  Is that what I need for an M0?  Or, do any other Arduino boards come with it built in (I notice the Zero for example also has a USB programming port)?

Any help with this would be appreciated.  Is there a good tutorial somewhere you would recommend?  Thanks again...

per1234, 2021-01-19:
>I kinda thought the SAMD boards, including the M0, came with the required circuitry.

Only the Zero and M0 Pro have a built-in debug probe. For any of the other official SAMD boards an external debug probe is required.

>I see the CMSIS probes can be purchased online and they appear to connect to a SMD port (which the M0 has - a 2x5 pin connector). Is that what I need for an M0?

Correct.

>Any help with this would be appreciated.

I'd be happy to help you out over on the [Arduino Forum](https://forum.arduino.cc/)

closed 2021-01-19 by per1234