arduino/arduino-pro-ide #413: Arduino Pro IDE crashes directly on start

fm-sys, 2021-02-02:
I've downloaded the latest release from the GitHub release page and extracted it to the Program Files directory. However, when trying to start the `Arduino Pro IDE.exe` it fails with the following exception:
[image]

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Version: arduino-pro-ide_0.1.3_Windows_64bit

kittaakos, 2021-02-04:
@fm-sys, thank you for letting us know.
 - Which version of IDE do you use?
 - How did you unpack the ZIP?

fm-sys, 2021-02-04:
>     * Which version of IDE do you use?

**0.1.3** downloaded from here: https://github.com/arduino/arduino-pro-ide/releases/tag/0.1.3 

>     * How did you unpack the ZIP?

The default zip extractor which is contained in windows, nothing special...

kittaakos, 2021-02-04:
Thank you for your help! 👍

> default zip extractor which is contained in windows, nothing special...

Indeed it can be relevant: #293. Too long paths during the extraction could cause malfunction. Can you please share the path to your installation?

By the way, does it work with the [nightly](https://github.com/arduino/arduino-pro-ide#nightly-builds) build? We have changed quite a few things since December.

Also, could you please check whether the file is missing or it's there? For me, it looks like this with the nightly, and the app starts without any problems:
```
c:\Users\kittaakos\Desktop\arduino-pro-ide_nightly-20210204_Windows_64bit\resources\app\scripts>dir
 Volume in drive C has no label.
 Volume Serial Number is 04F3-13CC

 Directory of c:\Users\kittaakos\Desktop\arduino-pro-ide_nightly-20210204_Windows_64bit\resources\app\scripts

02/04/2021  01:12 PM    <DIR>          .
02/04/2021  01:12 PM    <DIR>          ..
02/04/2021  04:21 AM               474 arduino-pro-ide-electron-main.js
02/04/2021  04:21 AM               786 notarize.js
               2 File(s)          1,260 bytes
               2 Dir(s)  380,572,643,328 bytes free

```

kittaakos, 2021-02-04:
> after that the IDE does at least start without problems

Thank you for the update. I am glad it works now on your side.

For the sake of transparency, I am going to keep the issue open until the next release is out.

fm-sys, 2021-02-10:
It's working fine with the new release. Thank you!

closed 2021-02-10 by fm-sys