arduino/arduino-pro-ide #414: Sketch Control button stops working after creating a new tab split

silvanocerza, 2021-02-03:
**Describe the bug**

Moving a tab to create a new split in the editor UI disables the Sketch Control button for some seconds.

**To Reproduce**
Steps to reproduce the behavior:
1. Create a new sketch
2. Create a new tab 
3. Drag the tab to create a new split
4. Click any of the two Sketch Control buttons

The menu won't appear, clicking again after some seconds will work.

**Expected behavior**

I'd expect the Sketch Control button to always work.

**Desktop (please complete the following information):**
 - OS: Windows and Linux
 - Version: `snapshot.d9c793e`

**Additional context**
None.

kittaakos, 2021-02-04:
> The menu won't appear, clicking again after some seconds will work.

Same as https://github.com/arduino/arduino-pro-ide/issues/335? Can you please check, @silvanocerza?

silvanocerza, 2021-02-04:
Yep, seems like it's the same bug, just tested. I'll close this one in favour of the old issue.

closed 2021-02-04 by silvanocerza