arduino/arduino-pro-ide #415: FILENAME.h file is not found.

sebasdt, 2021-02-05:
**Describe the bug**
the arduino pro ide can't find a local .h file that is in the same folder as in the main sketch file. while normal ide can.

audio.h is a file without a .cpp, i made it and consists lines of code for beat detection code. (i never had problems compiling this code)

[image]

[image]

**To Reproduce**
I don't know how I could reproduce. audio.h is as so far the only .h file that wouldn't get fetched by the pro ide.

**Expected behaviour**
the pro ide includes the local .h files without any problem

**Screenshots**
[image]

[image]

[image]

**Desktop (please complete the following information):**
 - OS: windows pro
 - Version:20H2

**Additional context**
I have made .h files so I could split up the code and have more oversight over the code. 
these .h files are in the same folder as the main sketch itself.

sorry if there's not enough data. just ask for more.

ubidefeo, 2021-02-05:
@sebasdt 
can you try the latest Nightly Build?

sebasdt, 2021-02-05:
yeah i can but where can i find it?
right now i use: arduino pro ide V0.1.3

per1234, 2021-02-05:
@sebasdt download links for the Arduino Pro IDE nightly build are here:
https://github.com/arduino/arduino-pro-ide#nightly-builds

sebasdt, 2021-02-05:
hi i'm back! it seems fixed!
but i get a interesting error back
No matching function for call to 'strcpy' note: candidate function not viable: requires 0 arguments, but 2 were provided
but it compiles and uploads without anyproblems

ubidefeo, 2021-02-05:
@sebasdt , can you try and rename your `.h` into `.ino`?
it should solve your issue.
We'll address full support for `.h` in the near future

sebasdt, 2021-02-05:
hi!

thanks for the help and the tip! 
it's solved for now!

closed 2021-02-05 by sebasdt