arduino/arduino-pro-ide #416: Opening projects with high file count causes issues (too many tabs)

ubidefeo, 2021-02-05:
**Describe the bug**
Projects such as Marlin 3D printer firmware contain a large amount of files.
To allow users to compile this in Arduino IDE (Marlin suggests to use the Makefile or PlatformIO for compilation) and not show the 1600+ tabs, most files have been moved into a subfolder named src.
When the IDE requests the file list to the CLI, it obtains a full list because of recursion.
Such recursion should not be permitted and only the content inside the root of the sketch folder should be returned.
On my machine it’s been opening the project for the past 15 minutes... and counting.

**To Reproduce**
Steps to reproduce the behavior:
1. Download Marlin 2 https://github.com/MarlinFirmware/Marlin
2. Copy the folder `Marlin` to your Sketchbook
3. Launch IDE Pro
4. Open the Marlin Sketch
5. Leave the house and come back next week 😁 
6. The tabs will keep opening

**Expected behavior**
Only 4 tabs should appear (the 4 supported files in the Sketch):
- `Marlin.ino`
- `Configuration.h`
- `Configuration_adv.h`
- `Version.h`

**Screenshots**

IDE Pro
[image]
IDE Java
[image]

**Desktop (please complete the following information):**
 - OS: Windows/Mac/Linux
 - Version: 0.1.3-nightly.20210204

**Additional context**
Opening an issue into the CLI. Link to be added

ubidefeo, 2021-02-05:
@kittaakos I have assigned this to you just for tracking reasons, but if the CLI returns the correct list in theory you won't have to do anything

kittaakos, 2021-02-05:
This issue is the duplicate of https://github.com/arduino/arduino-pro-ide/issues/373. It's a CLI issue.

kittaakos, 2021-02-05:
> but if the CLI returns the correct list in theory you won't have to do anything

Yes.

Closing, it's a duplicate.

closed 2021-02-05 by kittaakos