arduino/arduino-pro-ide #417: [ Preferences] Theme selection dropdown not updating until OK is pressed

ubidefeo, 2021-02-05:
**Describe the bug**
When opening the Preferences panel (on Windows) if a new theme is selected the dropdown item won't update its content.
Pressing OK will update the theme correctly.

**To Reproduce**
Steps to reproduce the behavior:
1. Open File > Preferences
2. Select the Theme dropdown menu
3. Select a Theme other than the currently applied
4. the dropdown item will remain unchanged

**Expected behavior**
The dropdown item will change according to the selected (new) item

**Screenshots**
[image]
[image]
[image]
[image]

**Desktop (please complete the following information):**
 - OS: Windows 10 Pro
 - Version: 0.1.3-nightly

kittaakos, 2021-02-05:
This is on purpose. We do not save anything unless you press OK. @ubidefeo, do you want to change the values right after editing before pressing OK? What is the desired behavior of Cancel? Thank you!

ubidefeo, 2021-02-06:
@kittaakos the problem is that when you select it you don't see what you have selected, which can be confusing.
Try changing the theme and see that you'll still have the last set in the dropdown so it seems I have not changed anything.
Once you press OK the theme changes

kittaakos, 2021-02-08:
> he problem is that when you select it you don't see what you have selected, which can be confusing.
> Try changing the theme and see that you'll still have the last set in the dropdown so it seems I have not changed anything.
> Once you press OK the theme changes

Thanks for your patience, @ubidefeo. Now, I see what you meant, it is indeed a bug. I am on it. 👍

kittaakos, 2021-02-08:
This should be fixed now.

https://user-images.githubusercontent.com/1405703/107233396-d7fc4a80-6a22-11eb-9592-085dd9a939c2.mp4