arduino/arduino-pro-ide #418: On Case-Insensitive FS, using libraries with broken Sketch name casing breaks the IDE at launch

ubidefeo, 2021-02-05:
**Describe the bug**
Only applies to Mac  OS with Case-Insensitive HFS+ / APFS or other case insensitive File Systems (basically nowhere else)

If one or more libraries contain Sketch examples whose names have case incompatibility with the containing folder,
the IDE breaks at launch while trying to populate the `File > Examples` menu.

**To Reproduce**
Steps to reproduce the behavior:
1. On Mac OS install the library `WiFiManager` or `M5Stack` for ESP32/8266
2. (re)launch the IDE
3. The process will be interrupted

**Expected behavior**
As it happens for Sketches, when the Sketch folder and the Sketch `.ino` files are not an exact match (in case) the Sketch should be ignored.
To test this, try opening a random sketch from your Sketchbook folder into your Finder, change the case of only one letter of either the Sketch folder name or the `.ino` file name and (re)launch the IDE.
This sketch will no more be available in your File>Sketchbook list

**Screenshots**
n/a

**Desktop (please complete the following information):**
 - OS: Mac OS Mojave with Case-Insensitive FS (default setting for legacy reasons)
 - Version: 0.1.3-nightly-20210204

**Additional context**
This is a CLI related issue, because CLI returns the sketch list to the gRPC interface.
My guess is that the Go Path module find the folder name, checks to see if a `.ino` file with the same name is contained and returns `true` wether or not the case of the Sketch `.ino` name is an exact match with the folder name.
The sketch is then assumed to be valid and made available to the IDE for menu item.
When the IDE tries to load the Sketch name it fails to move forward because it expects an exact name (case-sensitive)

ubidefeo, 2021-02-05:
@kittaakos also for this one, the solution will come from the CLI

kittaakos, 2021-02-11:
The fix is available as part of the `0.1.4` release. I am closing it.

closed 2021-02-11 by kittaakos