arduino/arduino-pro-ide #419: startup splash screen shows on all virtual desktop screens

sebasdt, 2021-02-06:
**Describe the bug**
This is more an enhancement thingy not really a bug. 
anyway, this "bug" happens when you switch virtual desktop screens and And the Arduino pro ide splash screen shows up.

**To Reproduce**
this only will happen when using a desktop or laptop with windows 10. (I only have windows)
1. startup Arduino pro ide
2. wait until the splash screen shows.
3. switch to another virtual desktop screen.

**Expected behaviour**
what I think needs to happen is:
show the splash screen on the virtual desktop screen where the Arduino pro ide was started.

**Screenshots**
https://user-images.githubusercontent.com/67417070/107114717-5cef3480-6868-11eb-8ae8-47ad3a67ab4d.mp4
[image]

**Desktop**
 - OS: windows 10 pro
 - Version: 20H2

**Additional context**
for me this isn't a problem, I can deal with it.

ubidefeo, 2021-02-06:
hey @sebasdt 
thank you so much for finding this quirk.
we would have probably not realised for a while 😬 

the bug label is good for us
low priority but will be addressed asap

ciao.ubi

sebasdt, 2021-02-06:
no problem man! 
glad that it has been found and good luck fixing this!
Since it will be fixed should I close this issue now?

ubidefeo, 2021-02-07:
@sebasdt please leave it open 
we'll close it once the bug is resolved and we do issue grooming :)