arduino/arduino-pro-ide #420: Can't upload to Curiosity Nano AVR128DA48

sstaub, 2021-02-08:
**Describe the bug**
Using the DxCore https://github.com/SpenceKonde/DxCore

The upload to the Curiosity Nano board work fine with Arduino IDE 1.8.13 on macOS Big Sur
but fails with the Arduino Pro IDE 0.13 nightly20210205
Upload error: Error: 2 UNKNOWN: uploading error: uploading error: exit status 1
It seems that the Pro IDE doesn't know the programmer. Maybe it is a problem that DxCore
use a new version of AVRDude which supports better upload with UDPI

**To Reproduce**
Steps to reproduce the behavior:
1. In Tools (see also screenshot)
- Programmer Atmel nEDGB
- Chip AVR128DA48 
2. Try to upload, compiling works
3. see screenshot

**Expected behavior**
Upload with nEDGB should work

**Screenshots**
[image]
[image]

**Desktop (please complete the following information):**
 - OS: macOS Big Sur 11.2
 - Version: 0.1.13 nightly20210205

per1234, 2021-02-08:
Hi @sstaub. You've rediscovered a bug you were the first to report last summer. You might recall this https://github.com/arduino/arduino-pro-ide/issues/309

The problem is that Arduino Pro IDE doesn't support this capability:
https://arduino.github.io/arduino-cli/dev/platform-specification/#upload-using-programmer-by-default

But not to worry, there is an easy fix: Just use **Sketch > Upload Using Programmer**.

closed 2021-02-08 by per1234

sstaub, 2021-02-08:
Ok, thank you . I forgot that the problem have fix.