arduino/arduino-pro-ide #424: Verbose upload output rate is very slow

per1234, 2021-02-08:
**Describe the bug**

The rate of the output from the upload tool shown when verbose upload output is enabled in the preferences is very slow. There seems to be some significant bottleneck in the communication stream between the tool and the Pro IDE's output pane.

**To Reproduce**
Steps to reproduce the behavior:
1. **File > Preferences**
2. Check the box next to "Show verbose output during: upload"
3. Click the "OK" button.
4. Upload a sketch to an AVR board, megaAVR board, or the "Programming Port" of a Zero. For some reason, I don't have the problem when uploading to a board that uses bossac as the upload tool (e.g., the "Native Serial" port of the Zero). The output from the upload tool slowly streams in to the output pane for seconds after the actual upload has finished.

**Expected behavior**
The upload tool output to be displayed in the output pane at approximately the same speed as when using it from the command line.

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Version: 0.1.4-snapshot.3042c1a
Date: 2021-02-08T10:04:20.854Z
CLI Version: 0.15.1 alpha [c7403ed2]

**Additional context**
I don't have any problems with the rate of the compilation output.

ubidefeo, 2021-02-11:
@per1234 did you happen to have your Serial Monitor open when this happened?
I am not experience such sluggish behaviour unless the SM is open

per1234, 2021-02-11:
>did you happen to have your Serial Monitor open when this happened?

No, I haven't used the Pro IDE's Serial Monitor at all yet. But it's certainly interesting information that you're finding a correlation.