arduino/arduino-pro-ide #425: Examples menu lacking "category"

ubidefeo, 2021-02-08:
**Describe the bug**
When opening the Examples submenu, the examples are listed with separating lines, these are supposed to have a category string
* Built-in Examples
* Examples for any board
* Examples for {CURRENTLY SELECTED BOARD}
* Examples from Custom Libraries

**To Reproduce**
Steps to reproduce the behavior:
1. Go to File > Examples

**Expected behavior**
These "labels" should be introduced.
The last one should read (mind the case) "Examples from custom libraries"
The Java IDE has this division

**Screenshots**
[image]
[image]

tigoe, 2021-02-24:
Also worth notuing that the Examples menu fails to scan the libraries for examples, as the IDE 1.8.x does. See @ubidefeo 's screenshots to see it.

kittaakos, 2021-02-24:
> Examples menu fails to scan the libraries for examples

Thank you for reporting it. I think it is off-topic.

I opened a follow-up issue with your remark: https://github.com/arduino/arduino-pro-ide/issues/442. Please comment there and help us figure out what problem did you find. Thanks!