arduino/arduino-pro-ide #430: Dirty exit dialog choice is ignored

per1234, 2021-02-10:
**Describe the bug**
When closing a sketch with unsaved changes, the user is prompted whether they want to exit without saving "Are you sure you want to close the sketch?":
[image]
If they click the "No" button, the exit occurs anyway and the unsaved changes are lost.

**To Reproduce**
Steps to reproduce the behavior:
1. **File > Preferences**
2. Uncheck the box next to "Auto save".
3. Click the "OK" button.
4. Make changes to a sketch without saving.
5. Close the sketch (by doing any of **File > Close**, **File > Quit**, or closing the window). The "Are you sure you want to close the sketch?" should be shown.
6. Click the "No" button. The sketch will close anyway and the unsaved changes will be lost.

**Expected behavior**
When the user clicks "No" on the "Are you sure you want to close the sketch?" dialog, the sketch should not be closed.

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Version: 0.1.4
Date: 2021-02-10T12:24:17.658Z
CLI Version: 0.15.2 alpha [0d4cb150]

kittaakos, 2021-02-11:
It works on macOS, but I could reproduce it on Windows. I need to verify it on Linux too.

kittaakos, 2021-02-11:
I have fixed and verified it on Windows. We will make this available to the public soon.