arduino/arduino-pro-ide #432: Unsaved changes are not compiled

per1234, 2021-02-11:
**Describe the bug**
The saved versoin of the sketch is always compiled, even when there are unsaved changes in the editor.

**To Reproduce**
Steps to reproduce the behavior:
1. **File > Preferences**
2. Uncheck the box next to "Auto save".
3. Click the "OK" button.
4. Open a sketch that will compile without errors (e.g., **File > New**).
4. Make changes to the sketch code that would cause a compilation failure. Don't save.
5. **Sketch > Verify/Compile**. The compilation passes.

**Expected behavior**
The sketch code in the editor should be compiled, as is done in the Arduino IDE.

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Version: 0.1.4
Date: 2021-02-10T12:24:17.658Z
CLI Version: 0.15.2 alpha [0d4cb150]

kittaakos, 2021-02-11:
> **Expected behavior**
> The sketch code in the editor should be compiled, as is done in the Arduino IDE.

I would argue with this. We either save automatically before running a compile, or we compile the previous, unsaved state. As a dev, I would never expect to compile something that is not saved.

per1234, 2021-02-11:
That decision was made 15 years ago, so there's no point in arguing about it.

ubidefeo, 2021-02-11:
@per1234 we'll make a task for @kittaakos :)

kittaakos, 2021-02-11:
> 6\. **Sketch > Verify/Compile**. The compilation passes.

@per1234, I assume the same should happen for upload using programmer. What about burning the bootload?  The bootloader is unrelated to the sketch state, right? Thanks!

per1234, 2021-02-11:
>I assume the same should happen for upload using programmer

Correct. Both "Upload" and "Upload Using Programmer" compile the sketch before doing the upload and they both currently compile the saved version of the sketch just as "Verify/Compile" does.

>What about burning the bootload? The bootloader is unrelated to the sketch state, right?

That's correct. The sketch is not compiled when doing a **Tools > Burn Bootloader**. It just goes straight to flashing a pre-compiled binary bundled with the board platform, typically using the same tool that is used for the upload process. So the open sketch is completely irrelevant to the Burn Bootloader process. The only things that affect it are the board selection, programmer selection, and the port selection (and the port is only used by a couple type of programmer).

kittaakos, 2021-02-11:
Thank you, Per. 👍 

I have implemented this, we will make it available to the public soon.