arduino/arduino-pro-ide #433: Starting debug session saves without warning

per1234, 2021-02-11:
**Describe the bug**
When you start a debug session with a dirty sketch, the sketch is automatically saved, even though the "Auto save" preference is disabled. Combined with the current behavior of compiling and uploading the saved version of the sketch (https://github.com/arduino/arduino-pro-ide/issues/432), this can be quite confusing.

**To Reproduce**
Steps to reproduce the behavior:
1. **File > Preferences**
2. Uncheck the box next to "Auto save".
3. Click the "OK" button.
4. Make a change to the sketch code. Don't save.
5. Click the "Start Debug" button. The dirty indicator disappears and the changes are saved to the sketch.

**Expected behavior**
Don't save the sketch when starting a debug session.

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Version: 0.1.4
Date: 2021-02-10T12:24:17.658Z
CLI Version: 0.15.2 alpha [0d4cb150]

kittaakos, 2021-02-11:
Unlike in #432, I can argue here as the behavior was not defined 15 years ago. This is how VS Code works.

per1234, 2021-02-11:
How about a save confirmation dialog?

kittaakos, 2021-02-11:
> How about a save confirmation dialog?

Sure. That's possible of course.

kittaakos, 2021-02-11:
Related: https://github.com/microsoft/vscode/issues/111850

ubidefeo, 2021-02-11:
@kittaakos we should directly save before every compile action.
So auto-save is good but it's up to the user to have it, but before compiling it should just do it

kittaakos, 2021-02-11:
> @kittaakos we should directly save before every compile action.
> So auto-save is good but it's up to the user to have it, but before compiling it should just do it

I am not sure if I can follow you, @ubidefeo. (Related: https://github.com/arduino/arduino-pro-ide/issues/432#issuecomment-777281286). This is about debug and not compile.

ubidefeo, 2021-02-11:
@kittaakos 
I'll extend on this in another issue :)