arduino/arduino-pro-ide #435: Support for external tools / plugins

szerwi, 2021-02-12:
**Describe the solution you'd like**
I would like to propose support for external tools / plugins which are compatible with Arduino IDE. Ex. https://github.com/me-no-dev/arduino-esp32fs-plugin
If it is already supported, how can I use tools in Pro IDE?
If it is not supported yet, are there any plans for implementing that?

ubidefeo, 2021-02-14:
@szerwi 
we have long had this in our list, but the pre-existing tools are built on Java and the new IDE is built on Electron.
In order to implement tools, we should offer entry points for those and developers should follow guidelines to write such tools in JS, or make them binary executables that could be launched.
We're not there yet, but this particular instance, the ESP exception decoder, is one of the most used tools out there :)

szerwi, 2021-02-14:
@ubidefeo Thanks for the reply. Do you mean that ESP exception decoder is one of the most used tools in Arduino IDE? Is there any way to run ESP exception decoder outside Arduino IDE and use it together with Pro IDE?

ubidefeo, 2021-02-14:
@szerwi we don't have any stats/numbers, there aren't many tools available, but I know that one is widely used to make sense of stack output dumped into the serial when the ESP crash.
You cannot use those tools outside of the Java IDE, they are made in Java and built to leverage the Tools system in the Java IDE