arduino/arduino-pro-ide #436: Arduino Pro IDE Helper attempting to connect to ds.services.visualstudio.com

sgreadly, 2021-02-15:
**Describe the bug**

Arduino is trying to connect/communicate with dc.services.visualstudio.com

This is on a clean install of 0.1.4 Alpha, and was not noticed in previous Pro versions. I have no extensions installed.

**To Reproduce**

Simply start Arduino Pro IDE 0.1.4 Alpha

**Expected behavior**

To not communicate with such websites - including ones owned by Microsoft.

**Screenshots**

[image]

**Desktop (please complete the following information):**

 - OS: MacOS Big Sur 11.2.1
 - Version: 0.1.4 Alpha

**Additional context**

Why is the IDE trying to connect to that URL? 
What data is being communicated or sent to them ?
This is potentially a security / privacy concern.

Ta.

ubidefeo, 2021-02-16:
@sgreadly 
We have included a VS Code extension for debugging which is not developed by us, 
and might have accidentally left its telemetry enabled.
We have no interest in sending telemetry data back to 3rd parties since we don't even collect it for our own use 😬 

Thank you so much for reporting this, we'll look into it and expand on this issue to explain how to disable it
ciao.ubi

sgreadly, 2021-02-16:
@ubidefeo 
Thanks! Good to know, and good info :) 

Happy to help.

ubidefeo, 2021-02-16:
@sgreadly 
can you disable Cortex Debug's Telemetry and test again?
In the bottom-left corner of the IDE you have a gear cog icon for the extended preferences.
Click it and choose "open preferences".
In the "Search Settings" field type "telemetry".
You'll be presented with an active checkbox for Cortex Debug's "Enable Telemetry".
Uncheck it, restart the IDE and see if it still tries to connect.

Keep me posted :)
u.

ubidefeo, 2021-02-16:
@kittaakos this setting should be disabled as default, can we take care of it?

sgreadly, 2021-02-16:
@ubidefeo 

I just disabled Telemetry & restarted (confirmed it's disabled), but that didn't seem to have stopped the request :/

ubidefeo, 2021-02-17:
hi @sgreadly 
We have a couple more extensions for JSON support that we added in our latest alpha and @kittaakos is looking at their telemetry status.
Sorry for the inconvenience, I'll let you know as soon as we track them down :)

kittaakos, 2021-02-17:
We introduced the change with:
```
ATL-835: Support for JSON file type. [24dfffa]
```

Here is the corresponding code from VS Code's JSON language feature (`1.46.1`):

https://github.com/microsoft/vscode/blob/1.46.1/extensions/json-language-features/client/src/jsonMain.ts#L227-L234

@sgreadly, this is the location where the VS Code extensions are stored in the IDE:

```
 % pwd   
/Applications/Arduino Pro IDE.app/Contents/Resources/app/plugins
 % ls -al
total 0
drwxr-xr-x@ 7 akos.kitta  admin  224 Feb 10 13:33 .
drwxr-xr-x@ 8 akos.kitta  admin  256 Feb 10 13:33 ..
drwxr-xr-x@ 5 akos.kitta  admin  160 Feb 10 13:33 cortex-debug
drwxr-xr-x@ 5 akos.kitta  admin  160 Feb 10 13:33 vscode-arduino-language-server
drwxr-xr-x@ 5 akos.kitta  admin  160 Feb 10 13:33 vscode-builtin-cpp
drwxr-xr-x@ 5 akos.kitta  admin  160 Feb 10 13:33 vscode-builtin-json
drwxr-xr-x@ 5 akos.kitta  admin  160 Feb 10 13:33 vscode-builtin-json-language-features
```

You can wipe `vscode-builtin-json` and `vscode-builtin-json-language-features`. Can you still reproduce it after the deletion of the two VS Code extensions? Note, after the removal, neither language features nor syntax highlighting will be supported for the JSON file-type.

sgreadly, 2021-02-17:
Thanks @kittaakos !

Yep that fixed it - no more requests after deleting those 2 plugins.