arduino/arduino-pro-ide #437: Windows 10 Pro Protenta Driver not found

kg117, 2021-02-16:
**Describe the bug**
Totally cleared PC of Arduino Classic and Pro IDEs: unable to get dpinst-amd64 to install Protenta Driver.
**To Reproduce**
Steps to reproduce the behavior:
C:\Users\kevin\.arduinocdt
C:\Users\kevin\.arduinoProIDE
C:\Users\kevin\AppData\Local\Arduino15
C:\Users\kevin\AppData\Roaming\Arduino Pro IDE         logs
C:\Users\kevin\.theia
C:\Users\kevin\AppData\Roaming\arduino-pro-ide
C:\Users\kevin\AppData\Roaming\.theia
===========================================
Uninstalled arduino-1.8.13-windows using Windows Apps
Cleared D:\Arduino\arduino-pro-ide_0.1.4_Windows_64bit
Cleared all folders in C:\Users\kevin listed above
Installed: https://downloads.arduino.cc/arduino-pro-ide/nightly/arduino-pro-ide_nightly-latest_Windows_64bit.zip
Plugged in Protenta H7 board.
Searched for Protenta Boards: None listed
IDE did not find port: Menu Tools->Port selected COM3
Dialogue appeared: Install mbed enabled boards? Yes/No : Yes : Completed
Dialogue appeared: Run dpinst-amd64: Windows App dialogue appeared: Do you wish to install? Yes but dpinst-am64 dialogue disappeared! No installation of Protenta Driver
"USB Serial Device (COM3)" appears in Device Manager
Properties: Microsoft 21/06/2006 10.0.19041.1
Cl;cked Drop-down list in IDE and selected "Arduino Portenta H7 (M7 core) at COM3"

Still no Arduino Driver available?

[folder location.txt](https://github.com/arduino/arduino-pro-ide/files/5986902/folder.location.txt)

[2021-02-16_log.log](https://github.com/arduino/arduino-pro-ide/files/5986904/2021-02-16_log.log)

[image]

**Desktop (please complete the following information):**

    OS: Windows 10 Pro
    Version: 19042.804

**Additional context**
Add any other context about the problem here.

ubidefeo, 2021-02-16:
@kg117 
>In Pro IDE Board Manager: Searched for Protenta Boards: None listed

if you search for "Portenta" it finds it, I'm afraid you searched for "Protenta" which has a typo in the letters "o" and "r" being inverted

@per1234 any advice for the driver issue?

per1234, 2021-02-16:
Closing as duplicate of https://github.com/arduino/arduino-pro-ide/issues/434

closed 2021-02-16 by per1234